MarlinFirmware/Marlin #24026: [BUG] Incorrect steps calculation for Markforged_XY/YX in mixed 0.9 and 1.8 degree stepper setup

rondlh, 2022-04-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm developing a MARKFORGED_XY system, and currently have a 0.9 degree stepper for Y, and 1.8 degree stepper for X. 
In configuration.h DEFAULT_AXIS_STEPS_PER_UNIT is set accordingly, and the X and Y movements move the correct amount. But moving Y also moves X because the amount of steps is calculated incorrectly.

I can solve this issue in planner.cpp
```
  #elif ENABLED(MARKFORGED_XY) // IRON >> 1
    block->steps.set(NUM_AXIS_LIST(ABS(da + db), ABS(db), ABS(dc), ABS(di), ABS(dj), ABS(dk), ABS(du), ABS(dv), ABS(dw)));
```
A dirty hack to solve this shows that it can work:
`    block->steps.set(NUM_AXIS_LIST(ABS(da + (db >> 1)), ABS(db), ABS(dc), ABS(di), ABS(dj), ABS(dk), ABS(du), ABS(dv), ABS(dw)));`
It seems that db was multiplied with the higher step count of the 0.9 degree stepper.

Perhaps this configuration is not supported, but it could be made to work quite easily (not with the described hack!).
I expect that the same issue occurs when using COREXY and it's variations.

### Bug Timeline

Don't know

### Expected behavior

Correct MARKFORGED movement, Y movement doesn't affect X position.

### Actual behavior

X position is changed with Y movement.

### Steps to Reproduce

1. Use 0.9 and 1.8 stepper in MARKFORGED setup, set DEFAULT_AXIS_STEPS_PER_UNIT accordingly.
2. Move Y, and see if X stays in the same place

### Version of Marlin Firmware

Marlin 2.0.9.3 latest bugfix

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### Add-ons

BTT TFT35 V3.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Other (explain below)

### Additional information & file uploads

BTT TFT35 V3.0 as host

thinkyhead, 2022-04-20:
Because the steps-per-mm value is configurable at runtime with `M92`, applying the ratio between the X and Y steppers will be a little extra taxing on the MCU, but of course it can be done. To reduce the amount of extra calculation required, we would recalculate that ratio any time the steps-per-mm are modified for one of the CORE / H-BOT steppers, whether that is by G-code, EEPROM load/reset, or the LCD menu. That way we only incur a couple of extra `float` multiplications in `Planner::_populate_block`.

rondlh, 2022-04-21:
OK, that's great input.
Please note that I don't use M92 to reconfigure anything. I just configure the DEFAULT_AXIS_STEPS_PER_UNIT in configuration.h according to the actual values, but the movement is incorrect.
I just realized that there might be an easy way for me to solve this issue. I could give the 1.8 degree stepper double the amount of micro steps than the 0.9 degree stepper, resulting in the same amount of steps for both X and Y. I guess (not tested yet, but will try it soon) that the math will work out again after this.

thinkyhead, 2022-04-21:
> I could give the 1.8 degree stepper double the amount of micro steps than the 0.9 degree stepper

Aha, I assumed you had already tried that. I hope it works!

rondlh, 2022-04-21:
> > I could give the 1.8 degree stepper double the amount of micro steps than the 0.9 degree stepper
> 
> Aha, I assumed you had already tried that. I hope it works!

I just thought of this micro steps option today, gave it a try and it works :D
I've build a dual carriage printer based on Markforged kinematics. The only thing that is currently not working is that the carriage that is not active doesn't compensate for the Y movement. I'm digging in the code to figure that out... any input would be appreciated.

rondlh, 2022-04-24:
Currently I recommend to use the same type of steppers for both X and Y of course, but adjusting the micro steps to have the same amount of steps per mm for both X and Y can be used to solve this issue.

closed 2022-04-24 by rondlh