MarlinFirmware/Marlin #24027: [FR] IDEX support for MARKFORGED_XY kinematics

rondlh, 2022-04-11:
### Is your feature request related to a problem? Please describe.

Currently Marlin does not support IDEX for MARKFORGED_XY kinematics.

### Are you looking for hardware support?

Nope

### Describe the feature you want

Independent dual X carriage (IDEX) support for the MARKFORGED_XY kinematics.

thisiskeithb, 2022-04-11:
Note: We’d need IDEX/CoreXYU support for standard CoreXY before this could be implemented: https://github.com/MarlinFirmware/Marlin/issues/6321

rondlh, 2022-04-11:
> Note: We’d need IDEX/CoreXYU support for standard CoreXY before this could be implemented: #6321

I don't know about that, to me it seems CoreXYU seems quite complicated compared to IDEX Markforged_XY. Here a drawing of how it looks:
[image].
I've build the system already, and got the carriages moving correctly with minor modifications, the key issue I currently have is that the carriage that is not selected doesn't compensate the X position for the Y movement.

rondlh, 2022-04-24:
I managed to "hack" Marlin to get this partially working for FULL_CONTROL and AUTO_PARK mode, using DUAL_X_CARRIAGE and Cartesian kinematics as a base configuration. The key issue I had was that the unused X stepper would not adjust for the Y movement. Still needs more work, currently I believe the speed calculations are probably incorrect, I will spy into the MARKFORGED_XY calculation to get this right and will look into duplication mode and mirrored mode too. I keep you posted on my progress...

rondlh, 2022-05-07:
Just a quick update. I managed to get most things working. Basically if you just enable DUAL_X_CARRIAGE and MARKFORGED_XY and disable the blocking line in SanityCheck.h
#error "DUAL_X_CARRIAGE cannot be used with COREXY, COREYX, COREXZ, COREZX, MARKFORGED_YX, or MARKFORGED_XY."
Then only 1 key issue remains, the unused carriage X position does not compensate for the Y movement. This can be solved in stepper.cpp
```
#elif ENABLED(DUAL_X_CARRIAGE) // IRON, implemented INVERT_X2_VS_X_DIR for DUAL_X_CARRIAGE, added 2x ^ ENABLED(INVERT_X2_VS_X_DIR) below
  #define X_APPLY_DIR(v,ALWAYS) do{ \
	if (extruder_duplication_enabled || ALWAYS) { X_DIR_WRITE(v); X2_DIR_WRITE((v) ^ ENABLED(INVERT_X2_VS_X_DIR) /*^ idex_mirrored_mode*/); } \
    else if (last_moved_extruder) X2_DIR_WRITE(v ^ ENABLED(INVERT_X2_VS_X_DIR)); else X_DIR_WRITE(v); \
  }while(0)
  #define X_APPLY_STEP(v,ALWAYS) do{ \
    if (extruder_duplication_enabled || ALWAYS) { X_STEP_WRITE(v); X2_STEP_WRITE(v); } \
    else if (last_moved_extruder) X2_STEP_WRITE(v); else X_STEP_WRITE(v); \
  }while(0)
#else
  #define X_APPLY_DIR(v,Q) X_DIR_WRITE(v)
  #define X_APPLY_STEP(v,Q) X_STEP_WRITE(v)
#endif
```
This update will do the following:
1. Send the needed X-position compensation steps to the unused carriage (T0/T1)
2. Enable INVERT_X2_VS_X_DIR for DUAL_X_CARRIAGE, I need to enable this to make things work
3. Enable DUPLICATION mode, but not MIRRORED duplication mode!!!
So in total MARKFORGED_XY in FULL_CONTROL, AUTO_PARK and DUPLICATION are working, but not MIRRORED duplication mode.
MIRRORED duplication mode will be quite difficult to implement, the reason is that the amount of steps required for the X2-stepper is not the same as required for the X-stepper in MIRRORED mode.

The current implementation of DUPLICATION and MIRRORED mode are as follows:
DUPLICATION mode:
1. Send  X and X2 to the correct start position
2. Send the same steps and direction signals to both X-steppers
MIRRORED mode:
1. Send X and X2 to the correct start position
2. Send the same steps, but invert the direction signals of the X2-stepper compared to the X-stepper.

IDEX is most suitable for system where X, Y and Z are independent, like in an i3 system.

mohamednaguib26w, 2023-03-08:
greetings
very good work 
but i have some issues
Can you help me?

rondlh, 2023-05-09:
Sure, talk... 
From my side I have abandoned this project, and went for an i3 IDEX approach.

rondlh, 2023-09-02:
Seems there is no interest in this

Daedalus7000, 2023-10-08:
Hello,
I also need some help with this issue.

I just completed a build of a printer with this kinematics (Markforged XY + Dual X Carriage) using EddieTheEngineer's FrankenVoron Tridex mechanics for the XY gantry and steppers, and using latest Marlin 2.1.2.1 firmware and the changes to stepper.cpp and SanityCheck.h listed above to drive the printer.
My kinematics looks like this:
 
[image]

The problem I have is when moving in Y direction, the compensation of the X stepper motors is the wrong direction. This means that instead of standing still in X direction, the hot-ends move outwards when moving the Y forward.

This is not helped by changing the direction of the X-stepper plugs or inverting the direction in Configuration.h, since when moving only the X stepper in Repetier Host with incresing numbers the hot-end moves towards X-max which is correct.

My thinking is that the problem is in stepper.cpp and the calculations for X-direction and X2-direction needs to be inverted when moving Y-stepper or Y2-stepper. But I'm not skilled enough to find where and what to modify.

Hopefully you can help me sort this out.

Cheers from Sweden.

thinkyhead, 2023-10-08:
> Seems there is no interest in this

This is a very active project with very limited resources. Please don't assume that there is no interest simply because we are busy with other things.

Daedalus7000, 2023-11-26:
Hello,
After a lot of testing, I think I'v solved this.

The first issue I had was that my printer was built as a mirrored version of an Markforged XY IDEX printer. Meaning the stepper for driving the (left) T0 carriage is positioned in the right back corner, and viseversa for T1. After I rebuilt the printer as a normal non-mirrored printer, I could start investigating.

Now the printer has the following kinematics:
[image]

Some note about my hardware setup. I'm using RAMPS 1.4 board and I'm using dual Y endstops located at the back (max) of the printer plugged in to Y_MIN and Y_MAX sockets on the board.
Then using Marlin-bugfix-2.1.x downloaded on November 4 2023, the following code snippet in Configuration_adv.h is critical to get it to work.

```
#if HAS_Y2_STEPPER
  //#define INVERT_Y2_VS_Y_DIR        // Y2 direction signal is the opposite of Y
  #define Y_DUAL_ENDSTOPS           // Y2 has its own endstop
  #if ENABLED(Y_DUAL_ENDSTOPS)
    #define Y2_STOP_PIN Y_MIN_PIN   // Y2 endstop pin override
    #define Y2_ENDSTOP_ADJUSTMENT  0  // Y2 offset relative to Y endstop
  #endif
#endif
```
Also, I'm NOT using INVERT_X2_VS_X_DIR as suggested by Rondlh,

Same as Rondlh, I have to disable the blocking line in SanityCheck.h
#error "DUAL_X_CARRIAGE cannot be used with COREXY, COREYX, COREXZ, COREZX, MARKFORGED_YX, or MARKFORGED_XY."

In Stepper.cpp the code for Dual_Y_Steppers need to be like this:
```
#if HAS_SYNCED_Y_STEPPERS
  #define Y_APPLY_DIR(FWD,Q) do{ Y_DIR_WRITE(FWD); Y2_DIR_WRITE(INVERT_DIR(Y2_VS_Y, FWD)); \
	if (!extruder_duplication_enabled) { \
		if (last_moved_extruder) \
			X_DIR_WRITE(FWD); \
		else X2_DIR_WRITE((FWD) ^ 1); \
	}; \
  }while(0)
  #if ENABLED(Y_DUAL_ENDSTOPS)
    #define Y_APPLY_STEP(FWD,Q) do { \
		DUAL_ENDSTOP_APPLY_STEP(Y,FWD); \
		if (!extruder_duplication_enabled) { \
			if (last_moved_extruder) \
				X_STEP_WRITE(FWD); \
			else X2_STEP_WRITE(FWD); \
		}; \
	}while(0)
  #else
    #define Y_APPLY_STEP(FWD,Q) do{ Y_STEP_WRITE(FWD); Y2_STEP_WRITE(FWD); }while(0)
  #endif
#elif HAS_Y_AXIS
  #define Y_APPLY_DIR(FWD,Q) Y_DIR_WRITE(FWD)
  #define Y_APPLY_STEP(FWD,Q) Y_STEP_WRITE(FWD)
#endif
```
Then the last issue can be solved in Planner.cpp by having different calculations for Extruder 0 and 1:
```
// Compute direction bit-mask for this block
...
#elif ENABLED(MARKFORGED_XY)
		if (extruder == 0) {
			dm.a = (dist.a + dist.b > 0);               // Motor A direction
			dm.b = (dist.b > 0);                        // Motor B direction
		}
		else {
			dm.a = (dist.a - dist.b > 0);               // Motor A direction
			dm.b = (dist.b> 0);                         // Motor B direction
		}
```

I have tested manual movements of both toolheads in Repetier Host in autoparking mode plus I have tested single colour prints with T0 toolhead. No problems there. I have not looked into Duplication Mode or Mirrored Duplication Mode yet.

I hope you all find this informational and usefull.

Cheers from Sweden.