MarlinFirmware/Marlin #24030: [BUG] Can't get in to the lcd menu

AfroPsycho, 2022-04-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hey, 
I am trying to hook up reprap discount smart controller to the skr 2.0 rev b
Everything is working fine it can detect sd, lcd is working fine I can rotate encoder to change feed rate from 100 to up or down reset button is functional 
But when I press to encoder button nothing is happening 
I can't get in to the menu 
I've checked in marlin if I accidentally uncomment no lcd menus or something but everything is fine
Encoder button was working fine with my previous mb 
I have also did continuity test on cable which is fine
I have tested the button between (+) and sw pins when I press to button it shows -5v when I release it it shows 0 volts 
But I still can't get in to the menu 

### Bug Timeline

When I changed my motherboard from mks gen 1.4 to skr 2.0 I have also compile a fresh marlin I was using 1.0.x now I am using 2.0.x

### Expected behavior

When pressed to the encoder button it should open the menu

### Actual behavior

Nothing is happening like button is not pressed

### Version of Marlin Firmware

Marlin 2.0.9.3

### Printer model

3D4E 4GenPro 2R ( Ultimaker clone made by Turkish company )

### Electronics

Upgraded board from stock mks gen 1.4 to BTT SKR 2.0 Rev B

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8472233/Configuration.zip)

rondlh, 2022-04-12:
Probably faulty/broken wiring or broken board. You can try to test the relevant pin manually and check the status with a multi meter and M43.

AfroPsycho, 2022-04-12:
Well it turns out manufacturer wired up encoder button different than normal 
the expansion pin used in mks board is not available in skr board 
So I have changed the wiring and problem solved

closed 2022-04-12 by AfroPsycho