MarlinFirmware/Marlin #24031: [BUG] Y_MIN not working on Trigorilla_14

GrevanGER, 2022-04-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Tried to configure Marlin 2.0.9 for my "Anycubic i3 Mega S", which runs a Trigorilla_14 board. 
Everything, incl. my BL_Touch seems to work fine, with the exception of the Y_Min endstop. 
The endstop switch itself on the i3 Mega, is one with a blue status LED, which works fine according to the LED itself
and my multimeter readings.

What I did so far: 
- Using Pronterface and the M119 gecode to check on the endstop status in the firmware. 
Result: "y_min: open" at all times
- Tracing down the wires of the endstop to check where they conect. 
Since the regular endstop pins are blocked by the adapter PCB which runs the LCD, Anycubic
has placed the Y_MIN on "D43". 
- Tried to add a line in the pins_TRIGORILLA_14.h
#if MB(TRIGORILLA_14)
  #define Y_MIN_Pin                           43
#endif
Since pins_RAMPS.h is included in the pins_TRIGORILLA_14.h, I went there and disable the 
AUX4_06_PIN which was defined to be pin 43. 

Result: No result

What am I doing wrong or what am I missing?

### Expected behavior

Expected the Y_Min to work as usual.

### Actual behavior

No response from Y_MIN at all.

### Steps to Reproduce

Install Marlin 2.0.9 on a i3 Mega S with a Trigorilla_14

### Version of Marlin Firmware

2.0.9

### Printer model

Anycubic i3 Mega S

### Electronics

TMC_2208 Steppers, BMG Extruder, BL_Touch (on Servo0)

rondlh, 2022-04-12:
You could give these a try in Configuration.h
Enable one of of them at a time.
#define ENDSTOPPULLUP_YMIN
or
#define ENDSTOPPULLDOWN_YMIN

Or use another digital IO for the Y_MIN. The blue LED could cause some issues here...

GrevanGER, 2022-04-12:
By default, ENDSTOPPULLUP was decleared for all endstops. 
So I have commented that out for the Y_MIN

Now the M119 reports Y_MIN to be TRIGGERED, no matter what I do. 

ENDSTOPPULLDOWN seems not to be an option. I got this error: 
#error "PULLDOWN pin mode is not available on the selected board."

ellensp, 2022-04-12:
Ramps pins are wrapped in #ifndef blocks
ie
```
#ifndef Y_STOP_PIN
  #ifndef Y_MIN_PIN
    #define Y_MIN_PIN                         14
  #endif

```

So ll you need to do is add #define Y_STOP_PIN 43 to Your Configuration.h 
I normally add these under #define MOTHERBOARD, so it obvious its that motherboard but with modified pins

closed 2022-04-12 by ellensp