MarlinFirmware/Marlin #24038: [BUG] After Printing, the display dont show correkt on stop/abort on ender 3v2 M2.0.9.3

gorillale, 2022-04-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After Printing, the display dont show correkt on stop/abort on ender 3v2 M2.0.9.3

### Bug Timeline

new on 2.0.9.3

### Version of Marlin Firmware

2.0.9.3

### Printer model

ender3v2

### Electronics

normal ender board 

### Add-ons

-

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Cura

### Additional information & file uploads

After Printing, the display dont show correkt on stop/abort on ender 3v2 M2.0.9.3

Bob-the-Kuhn, 2022-04-14:
We need more information before we can be of help.

A picture of the display along with a description of the problem(s) on the display is a must.  Best if we could see a good and a bad display.

Some questions that come to mind:

- What version of Marlin worked correctly?
- How is the print stopping?  Normal completion, sending abort command, removing SD card, ... ?

Your configuration files are also usually needed.

ellensp, 2022-04-14:
Also need to identify the model of e3v2 display.
There are now 3 different known models of this display type.
Needs an image of the inside of the display

gorillale, 2022-04-15:
Hi, pleas look at my picture on Marlin 2.0.9.3 (before 2.0.9.2 other display when printing)
I updated the display to from marlin folder after compile marlin.
Not known what chip you mean, i have the 4.2.2 board not the 4.2.7, the printing is working before, but on marlin 2.0.9.2 i not have the leveling hint of all points to display after leveling .
The second problem i have see to make this photos. 
- start printing on fresh started printer, (not homed).
- abort print, hotbeed move to the edge an overrun no detekt the endswitch?
- dont move the bed when not homed, it is my idea.

now the photos:
[image]
[image]
[image]
[image]
[image]

gorillale, 2022-04-15:
and the display board with unnamed chip:
[image]
[image]

gorillale, 2022-04-15:
i hope this can help to find a solution,
thank you for your working on marlin.
regards
andrew

thinkyhead, 2022-04-20:
> … `DWIN_CREALITY_LCD_ENHANCED` …

It is now called `DWIN_LCD_PROUI`.

gorillale, 2022-04-20:
Sorry, I not have a entry with "DWIN_LCD_PROUI"

ellensp, 2022-04-23:
@gorillale then your not testing bugfix https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration.h#L2931

gorillale, 2022-04-23:
Ist this not Creality Display? I have buy an original ender 3 v2 with 4.2.2 Board inside.
Why DWIN_LCD_PROUI  ?

gorillale, 2022-04-24:
i no more can compile the bugfix version:
Error: Build environment 'STM32F103RET6_creality' is incompatible with BOARD_CREALITY_V4. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple

I no more sse what is to do, no manual no instructions.... i think i switch to original ware from creality, no future for me... i no a developer and without a seriosli manual to install all this i not can use this...

thank you, happi programing

ellensp, 2022-04-24:
The valid options are the following as listed in the error.

STM32F103RE_creality
STM32F103RE_creality_xfer
STM32F103RC_creality
STM32F103RC_creality_xfer
STM32F103RE_creality_maple
etc

STM32F103RET6_creality is no longer valid option replaced with STM32F103RE_creality

mriscoc, 2022-04-24:
I you don't have the definition of DWIN_LCD_PROUI you must try with the latest bugfix.

thisiskeithb, 2022-04-24:
@gorillale: Please start with a fresh copy of [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) since you've mixed up files/versions.

gorillale, 2022-04-24:
Thank you all, the STM32F103RE_creality realy works, but other errors missing function i see.
I have downloaded marlin bugfix2 zip and marlin config bugfix2 zip.
copied the config from creality ender 3 v2 to the marlin folder
activated my settings bl touch.
i make this with all version before, but now many errors with missing functions
not known what i doing, is make this from a manual how to compile marlin with 
visual studio code

i will try moring the entry procedure again.

gorillale, 2022-05-02:
It is not compilable, i have downloaded marlin and bugfixes.
Edited the Ender3v2 header to set bl-touch, now compiled.
Many Errors and warnings will be displayed, normal creality display compile is ok.
with xx_PROUI missing functions and errors.
simply uncompileable.

ellensp, 2022-05-02:
@gorillale You expect to much.

The only User interface for this display that supports all features is DWIN_MARLINUI.

All other user interfaces have specific requirement in what feature they will or will not support. 
The designers of these other user interfaces only implement support for the features they want.
They are not all drop in replacements for each other.

In saying that 
to convert from DWIN_CREALITY_LCD to DWIN_LCD_PROUI for a standard Ender 3v2 only requires the following changes

Marlin even tells you what you need to change

First error
#error "INDIVIDUAL_AXIS_HOMING_SUBMENU is required with ProUI."
You find INDIVIDUAL_AXIS_HOMING_SUBMENU In Configuration.h and enable it

Second error
#error "LCD_SET_PROGRESS_MANUALLY is required with ProUI."
You find LCD_SET_PROGRESS_MANUALLY in Configuration_adv.h and enable it

Third error 
#error "STATUS_MESSAGE_SCROLLING is required with ProUI."
You find STATUS_MESSAGE_SCROLLING in Configuration_adv.h and enable it

Forth error
#error "BAUD_RATE_GCODE is required with ProUI."
You find BAUD_RATE_GCODE in Configuration.h and enable it

Fifth error
#error "SOUND_MENU_ITEM is required with ProUI."
You find SOUND_MENU_ITEM in Configuration_adv.h and enable it

Last error
#error "PRINTCOUNTER is required with ProUI."
You find PRINTCOUNTER in Configuration.h and enable it

These error message are there to help you, why do you ignore them?

Here are current bugfix configs for a Ender 3v2 with DWIN_LCD_PROUI a on a BOARD_CREALITY_V4 controller

[Bufix configuration Ender3v2 V422.zip](https://github.com/MarlinFirmware/Marlin/files/8602433/Bufix.configuration.Ender3v2.V422.zip)

gorillale, 2022-05-02:
@ellensp Very thank you for your help, i not thinking that is into the Configuration_adv.h, sorry
Now i try again the same procedure.
Regards.

gorillale, 2022-05-03:
compile is working, but repetier can connect no more.
i wondering what your doing, bug fix changed menus connection etc...
i think bether for me to switch original creality software that is 
teste and run without bugs.
(hint: Baudrate is 115k, same on repetier)

Thank you and happy programing

closed 2022-07-14 by github-actions[bot]