MarlinFirmware/Marlin #24039: [FR] (feature summary)

jj57, 2022-04-14:
### Is your feature request related to a problem? Please describe.

Hello ! I've downloaded your firmware update V4.2.7 on this site : [https://www.ender5.fr/carte-mere-creality-v4-2-7/](url) for my Creality 5 with BRTouch Creality. The problem is that the "Z" axis is mooving doubling the selected high in Cura :-(
A friend of mine told me that's coming because the first Ender 5 had a 400 steps endless screw for "Z" axis and the next a 800 steps for this axis.
Do you think it could be possible to change this in the firmware ?
Thanks a lot for your answer.
Best regards
Jean-Jacques from France

### Describe the feature you want

Reducing the speed levelling in "Z" axis of my Ender 5

### Additional context

[image]
[image]

ellensp, 2022-04-14:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-04-14 by ellensp

jj57, 2022-04-14:
Thanks a lot for the very fast answer.
I'm gonna do what you recommanded.