MarlinFirmware/Marlin #24040: [BUG] Issue with MULTI_VOLUME

Guilouz, 2022-04-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I think there is issue with new STM32F429VGT6 MCU (embedded in new SKR 2) and MULTI_VOLUME support.

When MULTI_VOLUME and USB_FLASH_DRIVE_SUPPORT are enabled, motherboard reboots exactly every 3min when a microSD card is inserted into the slot before booting.
If no microSD is inserted at startup, no issue. If microSD is inserted after booting, no issue.

When MULTI_VOLUME and USB_FLASH_DRIVE_SUPPORT are disabled all scenarios work without issue.

This issue does not occur with SKR 2 equipped with STM32F407VGT6 MCUs.

### Version of Marlin Firmware

Latest bugfix

### Electronics

BigTreeTech SKR 2

### Additional information & file uploads

[Config_Files.zip](https://github.com/MarlinFirmware/Marlin/files/8492193/Config_Files.zip)

Guilouz, 2022-04-15:
Other issues report this too:

https://github.com/bigtreetech/SKR-2/issues/111

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/610

https://github.com/bigtreetech/SKR-2/issues/109

thisiskeithb, 2022-04-15:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23618

thisiskeithb, 2022-04-15:
There's no need to tag us. Someone will look at #23618 when they can.

closed 2022-04-15 by thisiskeithb