MarlinFirmware/Marlin #24041: [BUG] incorrect z movements with arcs

flaviut, 2022-04-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm slicing this part in vase mode in Cura, with [Arc Welder](https://github.com/FormerLurker/ArcWelderLib) enabled and "Allow 3d Arcs" on. Turning 3d arcs off works fine.

Here's a video. At first, it prints back & forth around the circle without a problem (or maybe just a little bit of a problem). Then it switches into going all the way around continuously, but the z-axis does not move. Eventually it decides to move the Z-axis 2-3mm, all at once.

I looked through the g-code, and I don't see any rapid z-axis moves like this.

https://user-images.githubusercontent.com/5213469/163416610-50612416-52d0-4408-b6b7-c6ed366db945.mp4

### Bug Timeline

Unknown, I've on used bugfix-2.0.x

### Expected behavior

I expected it to print smoothly in vase mode.

### Version of Marlin Firmware

29d3f6a024672a960f7e033e360178f4e80ce98b

### Printer model

Ender 5

### Electronics

SKR2

### Additional information & file uploads

config: https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...flaviut:ender5-skr2-bugfix-2.0.x

log output from around the right place:
<details>
```
Recv: ok N1042 P0 B2
Send: N1056 G3 X120.578 Y98.430 Z4.107 I-3.951 J15.283 E299.15126*37
Recv: echo:N1043 G2 X119.255 Y123.242 Z3.906 I27.314 J25.938 E295.30732*14
Recv: ok N1043 P0 B2
Send: N1057 G3 X124.946 Y102.407 Z4.123 I-198.887 J222.828 E299.46074*37
Recv: echo:N1044 G1 X117.888 Y124.749 Z3.912 E295.41391*12
Recv: ok N1044 P0 B2
Send: N1058 G3 X126.477 Y107.529 Z4.138 I-4.076 J4.008 E299.75215*46
Recv: echo:N1045 G3 X115.357 Y126.305 Z3.920 I-4.144 J-3.905 E295.57138*4
Recv: ok N1045 P0 B2
Send: N1059 G3 X125.175 Y113.912 Z4.156 I-266.760 J-51.089 E300.09344*63
Recv: echo:N1046 G3 X112.482 Y126.407 Z3.928 I-1.647 J-5.845 E295.72357*12
Recv: ok N1046 P0 B2
Send: N1060 G3 X121.568 Y120.577 Z4.177 I-15.192 J-3.913 E300.49437*50
Recv: echo:N1047 G3 X106.087 Y125.166 Z3.946 I34.633 J-195.565 E296.06487*21
Recv: ok N1047 P0 B2
Send: N1061 G3 X117.362 Y125.142 Z4.194 I-156.046 J-139.554 E300.81959*8
Recv: echo:N1048 G3 X99.419 Y121.573 Z3.967 I4.032 J-15.467 E296.46544*45
Recv: ok N1048 P0 B2
Send: N1062 G3 X115.362 Y126.303 Z4.200 I-3.842 J-4.314 E300.94163*6
Recv: echo:N1049 G3 X95.023 Y117.626 Z3.983 I192.530 J-218.853 E296.77496*16
Recv: ok N1049 P0 B2
Send: N1063 G3 X112.149 Y126.437 Z4.209 I-1.844 J-5.625 E301.11225*13
Recv: echo:N1050 G3 X93.570 Y112.478 Z3.998 I4.146 J-3.949 E297.06637*28
Recv: ok N1050 P0 B2
Send: N1064 G3 X105.583 Y125.037 Z4.227 I23.128 J-124.567 E301.46401*30
Recv: echo:N1051 G3 X94.830 Y106.089 Z4.016 I234.577 J42.943 E297.40754*7
Recv: ok N1051 P0 B2
Send: N1065 G3 X99.219 Y121.381 Z4.247 I4.464 J-15.138 E301.85201*47
[... 250 lines cut ...]
Recv: ok N1133 P0 B2
Send: N1147 G3 X103.839 Y94.350 Z5.456 I147.089 J155.020 E325.05513*59
Recv: echo:N1134 G3 X124.861 Y114.972 Z5.278 I-265.307 J-71.364 E321.63775*58
Recv: ok N1134 P0 B2
Send: N1148 G3 X108.886 Y93.362 Z5.471 I3.511 J4.551 E325.33438*54
Recv: echo:N1135 G3 X120.594 Y121.571 Z5.300 I-15.116 J-5.095 E322.05362*55
Recv: ok N1135 P0 B2
Send: N1149 G3 X115.315 Y95.263 Z5.489 I-46.389 J168.704 E325.68563*40
Recv: echo:N1136 G1 X118.736 Y123.321 Z5.307 E322.18733*5
Recv: ok N1136 P0 B2
Send: N1150 G3 X121.560 Y99.410 Z5.510 I-5.394 J14.899 E326.08212*34
Recv: echo:N1137 G1 X118.54 Y123.523 E322.20207*111
Recv: ok N1137 P0 B3
Send: N1151 G3 X125.632 Y103.852 Z5.526 I-114.448 J109.002 E326.39784*38
Recv: echo:N1138 G3 X116.275 Y125.623 Z5.316 I-23.910 J-23.517 E322.36391*12
Recv: ok N1138 P1 B2
Send: N1152 G3 X126.660 Y108.892 Z5.541 I-4.523 J3.547 E326.67715*39
Recv: echo:N1139 G3 X114.985 Y126.379 Z5.320 I-3.429 J-4.373 E322.44246*13
Recv: ok N1139 P0 B2
Send: N1153 G3 X124.545 Y115.835 Z5.560 I-120.044 J-32.775 E327.05748*48
Recv: echo:N1140 G3 X111.125 Y126.593 Z5.331 I-2.233 J-5.356 E322.64885*11
```
</details>

GCode: [gcode.zip](https://github.com/MarlinFirmware/Marlin/files/8486254/gcode.zip)

ellensp, 2022-04-15:
"Allow 3D Arcs
This option allows G2/G3 commands to be generated when using vase mode. This is an experimental option, and it's possible that there are some unknown firmware issues when adding Z coordinates to arc commands. That being said, I've gotten pretty good results from this option. At some point, this will be enabled by default.

Type: Flag
Default: Disabled
Short Parameter: -z
Long Parameter: --allow-3d-arcs
Example: ArcWelder "C:\thing.gcode" --allow-3d-arcs"

As far as I'm aware Marlin does not support 3d arcs

ManuelMcLure, 2022-04-15:
An arc with a Z component should work per the code in `plan_arc`:
```Plan an arc in 2 dimensions, with linear motion in the other axes.```
So if this is not working it’s a bug.

ManuelMcLure, 2022-04-15:
That said, you should examine the gcode to make sure the G2/G3 commands are as expected. This might be an ArcWelder bug.

ellensp, 2022-04-15:
It not a bug when it is not implemented feature. 
Official gcode specification  https://reprap.org/wiki/G-code#G2_.26_G3:_Controlled_Arc_Move
No Z component

ManuelMcLure, 2022-04-15:
Yes, but Marlin _does_ implement it.

DerAndere1, 2022-04-16:
Marlin documentation already mentions the Z component: https://marlinfw.org/docs/gcode/G002-G003.html

The Z axis, is mapped to `axis_l` in G2_G3.cpp. Last time I tested it, it worked as expected with additional axes. 

@flaviut : Please attach your configs.

DerAndere1, 2022-04-21:
@flaviut please test the gcode script attached to this comment. It is an excerpt from the example gcode you provided with all extrusion removed and feedrate reduced. It should home, then move the toolhead in an oval helix, starting at hight Z1.964 and ending at Z2.240 . On my machine (`BOARD_ANET_10`, A4988 stepper drivers) it works without issues.
[CB_issue_24041_test1.zip](https://github.com/MarlinFirmware/Marlin/files/8535715/CB_issue_24041_test1.zip)

DerAndere1, 2022-04-22:
Does the issue also occur if you disable the option `ENABLE_LEVELING_FADE_HEIGHT` ?

flaviut, 2022-04-22:
My printer is in bits right now :)

I'll take a look and let you know as soon as it's back together

jars28, 2022-06-17:
I did a bit of testing. Will test the script when I'm back at my printer. I made a log of my attempts and added comments.
[marlin 2.1.x console session w notes.txt](https://github.com/MarlinFirmware/Marlin/files/8927544/marlin.2.1.x.console.session.w.notes.txt)
.

DerAndere1, 2022-07-07:
In the test script are you sure that the following sequence should result in any movement? I don't think so:
```
Send: G91
Recv: ok
[...]
Send: G18
Recv: ok
[...]

//HAPPY LIL VERT CIRCLE
Send: G2 X5 Z5 I5
```
Try this instead (I get a vertical arc with simultaneous z movement (i.e. helical movement, as expected):
```
G91
G18
G2 X5 Z5 I4
```

From the next command in your test script I also don't expect any movement: 
```
G2 X5 Y5 Z5 I5
```
Try this instead ( I get a vertical arc with simultaneous movement in Z, resulting in helical movement, as expected):
```
G2 X5 Y5 Z5 I4
```

jars28, 2022-07-16:
It moved in a vertical circle in the X and Z directions, with G91 active and after G18 was entered, but when I added a Y coordinate, there was no movement in the Y direction. My expectation from the documentation on G2 and G3 was that it would calculate and segment the linear Y positions along with calculating the positions along the arc for the two relevant axis(es?). 

The entire function looks like it has been rewritten in the bugfix version since I tested it, so I will try again when I am at the office with the printer and down time and not out in the field.

DerAndere1, 2022-07-16:
I confused G18 and M18 and forgot to enable CNC_WORKSPACE_PLANES. Padon me. I will redo the test, too

DerAndere1, 2022-07-25:
With `CNC_WORKSPACE_PLANES` enabled, I did not observe any unexpected behaviour with the test from https://github.com/MarlinFirmware/Marlin/issues/24041#issuecomment-1158838657. I.e., where the comment in the script says "HAPPY LIL VERT CIRCLE, Y DOESN'T MOVE AT ALL", I actually observed a helical movement (vertical circle in XZ plane with simultaneous movement of the Y axis). Where the comments in the script say "TINY ARC, NO Z MOVE", I observed movement of Z for the small distance of 0.1 mm. Again, I am testing with all bed leveling disabled in my configs

closed 2022-10-05 by github-actions[bot]