MarlinFirmware/Marlin #24043: Marlin latest bug fix update no temp. readings from hotend.

SamppaD78, 2022-04-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Old version of Marlin 2.9.3 with dyze high temp thermistor,temp table 66 is working,temperature is detected on the hotend . Updated today to bug fix  2.0 exactly the same marlin config as 2.9.3 version but no readings from hotend thermistor at all.. Displays 0C/0C. Only after reversing to old firmware i get readings from thermistor. To be clear as this is a hi-temp thermistor TD500 the room temperature values are not stable ,this was always issue with 32bit boards and this thermistore. Hoewer thermistor is mostly stable with higher temperatures.

### Bug Timeline

New issue,started after updating to bug fix 2.0

### Expected behavior

Expected that firmware displays normal temperature readings from hotend

### Actual behavior

After update to bugfix2.0 no reading from hotend. Room temp reading show 0C if I try to preheat hotend reading is 0C and after that thermal safety is triggered and printer is halted.
Downgrading firmware to 2.9.3 resolve this issue.

### Steps to Reproduce

Update to Marlin bugfix 2.0X CONFIGURATION_ADV_H_VERSION 02010000 
Use temp sensor table number 66 for DT500 dyze sensor
Compile the firmware and upload.
Reboot and check temp readings on hotend.
Observed reading is 0C on room temp value and 0C on hotend with preheating turned on
After this thermal runaway safety is triggered
----------------------------------------------------------------------------------------------------------------------------------------------
Disabled futures
Junction Deviation,Linear advance,Arc future,Square wave stepping,Hybrid threshold.
Enabled futures:
Stelthchop,S curve,Classic jerk,Power loss recovery,Host command,Cancel object,PSU control,TMC debbug,Pins debbug.

### Version of Marlin Firmware

Latest bugfix 2.0X

### Printer model

Ender 3 pro

### Electronics

SKR mini E3 2.0 STM32 2209

### Add-ons

Removed all hardware add-ons  for testing PSU switch,UPS

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

Issue is within the bugfix 2.0 code. Downgrading to 2.9.3 non bugfix version of marlin resolve this problem. All config settings within firmwares 2.9.3 and bugfix are the same but update outcome is diffrend as bugfix displays no readings at all on hotend.
[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/8495420/Config.files.zip)

ellensp, 2022-04-15:
Many build environment options available for this controller board, what of the following did you try?

STM32F103RC_btt
STM32F103RC_btt_USB
STM32F103RE_btt
STM32F103RE_btt_USB 
STM32F103RC_btt_maple 
STM32F103RC_btt_USB_maple 
STM32F103RE_btt_maple
STM32F103RE_btt_USB_maple

SamppaD78, 2022-04-15:
Compiled using STM32F103RC_btt.

thisiskeithb, 2022-04-15:
I’m wondering if there are multiple vendors trying to use this thermistor number in firmware when they should be using a different one. There’s another related issue with temps being off: https://github.com/MarlinFirmware/Marlin/issues/23737

You can try reverting this change since it was made after 2.0.9.3 was tagged: https://github.com/MarlinFirmware/Marlin/pull/23585

SamppaD78, 2022-04-15:
Using 2.9.3 solve this issue,but it would be nice to know the reason for this as a bugfix2.0 implement solutions for other problems. So update to bugfix 2.0 would be a good idea at some point of time.

ellensp, 2022-04-15:
Odd. 
I checked type 66 on ramps, temperatures are read and changing as expected (using variable resistor)
I also checked type 66 on a BOARD_BTT_SKR_E3_DIP  (same build environment STM32F103RC_btt) temperatures are read and changing as expected (using variable resistor)

Sadly  I don't have a actual  BOARD_BTT_SKR_MINI_E3_V2_0 to try.

thisiskeithb, 2022-04-15:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23737

thisiskeithb, 2022-04-15:
I didn’t realize OP opened multiple issues.

closed 2022-04-15 by thisiskeithb

SamppaD78, 2022-04-15:
Yes. I don't have but skr mini e3 2.0. I tested multiple times, compiling marlin from scratch, disabling Marlin futures,and only link to get this issue is Trianglelabs Dyze DT500 temp sensor on a temp sensor table 66 with skr mini and with bugfix2.0. When testing noticed that if temp table is set to 1 on purpose for  this dyze design temp sensor it will report error on screen with bugfix installed,if set to 66 no readings.. When i set same incorrect value 1 or correct value 66 with marlin 2.9.3 it reports normal temperature readings no errors regardless of settings!? Strange thing,as i would expect,error or incorrect temperature reading due to incorrect firmware temp sensor table configuration regardless of firmware version.