MarlinFirmware/Marlin #24050: BLTouch CRTouch ABL does not halt if probe is not triggered when missing the bed while probing before print

DiskDoctor, 2022-04-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When probing before print, if probe misses bed after being deployed and is not triggered by the bed surface, Printer is not halted and continues to attempt print are far too low nozzle, gouging into bed surface. 

Printer physical configuration problem caused the condition. Y-axis endstop found to be too far back, allowing bed to travel beyond proper limit.

### Bug Timeline

unknown

### Expected behavior

If probe is deployed but is not triggered by encountering a surface (at any location really), printer should halt and provide error feedback to user

### Actual behavior

No error is reported and print continues at much too low nozzle position, gouging the bed.

### Steps to Reproduce

Reproduce error by moving Y-axis endstop too far to rear, allowing bed travel beyond surface.

Begin print when pre-print probing misses front left corner of bed

### Version of Marlin Firmware

2.0.9.3 customized

### Printer model

Creality Ender 3 Pro, 1 week old

### Electronics

4.2.2 with A4988, STM32F103 512k

### Add-ons

CR Touch

### Bed Leveling

ABL Bilinear mesh

### Host Software

SD Card (headless)

### Additional information & file uploads

Missing error check when probing pre-print and probe misses bed
Considered SEVERE as physical damage can occur to print bed
Using pre-sliced, vendor provided 'dog'  No enduser slicer involved

thisiskeithb, 2022-04-16:
You didn't include your configs, but you need to enable `G29_RETRY_AND_RECOVER` so `G29_HALT_ON_FAILURE` will work: https://github.com/MarlinFirmware/Marlin/blob/5eeb6e0bcafe21d6d7169cf5c1689b199909aed7/Marlin/Configuration_adv.h#L2103-L2110

closed 2022-04-16 by thisiskeithb

DiskDoctor, 2022-04-16:
Thank you. This should be default behavior, but you've solved it for me for now.

thinkyhead, 2022-04-20:
> This does NOT RESTART the entire probing sequence, just the current probe position, right?

It might work that way with UBL. But for other leveling systems there is no resume once they have exited because preserving the state of the last `G29`-in-progress comes with a cost in both Flash and SRAM.

DiskDoctor, 2022-04-29:
Thanks for the clarification. I do suggest the halt on failure be enabled by default. Not so much a Bug, but "better"