MarlinFirmware/Marlin #24053: [BUG] Move from ErikZalm to Marling Firmware broke Github notifications

drf5n, 2022-04-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin works fine, but the move from https://github.com/ErkkZalm/Marlin/issues/ https://github.com/MarlinFirmware/Marlin/issues/ seems to have confused something in Github Notifications.

I've been getting emails from github about a number of old, closed issues, and if I look at my Github  Notifications screen , several of them had broken links, like:

  https://github.com/ErikZalm/Marlin/issues/701#issuecomment-1100882945

...which would not clear the notification.

I found a workaround, which was to manually edit the  404'd URL to be:

 https://github.com/MarlinFirmware/Marlin/issues/701#issuecomment-1100882945

and then it would clear out of my notifications queue.

### Bug Timeline

Not sure

### Expected behavior

I expected old, closed issues to not cause new notifications, and if they did, they wouldn't lead to 404s

### Actual behavior

I've been getting emails from github about a number of old, closed issues, and if I look at my Github  Notifications screen , several of them had broken links, like:

  https://github.com/ErikZalm/Marlin/issues/701#issuecomment-1100882945

...which would not clear the notification.

### Steps to Reproduce

1) Click on the Github notifications icon,
2) Click on a Marlin Notification
3) get a 404
4) go back, refresh, repeat...

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

RepRap Wallace

### Electronics

Teensy++

### Your Slicer

Slic3r

### Host Software

Pronterface

### Additional information & file uploads

It isn't a Marlin bug, so much as some Marlin+Github Migration/Moving/Administration weirdness.

thisiskeithb, 2022-04-17:
There is a bot running that is locking old, closed issues. It runs in batches each day and should be done sometime soon.

closed 2022-04-17 by thisiskeithb

thisiskeithb, 2022-04-17:
Feel free to file a bug report with github, but there is nothing actionable on our end: https://support.github.com/contact/bug-report

cp2004, 2022-04-17:
You could temporarily remove the 'issue locked' comment, which would mean users were not notified when these old issues were locked, and then up the pace (run each hour or something). I don't watch this repo, but I guess for those that do it is probably getting a bit annoying? I'm just looking out for you 🙂 . That's what was done for OctoPrint to lock old issues I think.