MarlinFirmware/Marlin #24054: DGUS display & Preheat

martindecz, 2022-04-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In Conditionals_post.h
`#if HAS_TEMPERATURE && EITHER(HAS_MARLINUI_MENU, HAS_DWIN_E3V2)
  #ifdef PREHEAT_6_LABEL
    #define PREHEAT_COUNT 6
  #elif defined(PREHEAT_5_LABEL)
    #define PREHEAT_COUNT 5
  #elif defined(PREHEAT_4_LABEL)
    #define PREHEAT_COUNT 4
  #elif defined(PREHEAT_3_LABEL)
    #define PREHEAT_COUNT 3
  #elif defined(PREHEAT_2_LABEL)
    #define PREHEAT_COUNT 2
  #elif defined(PREHEAT_1_LABEL)
    #define PREHEAT_COUNT 1
  #endif
  #if PREHEAT_COUNT && ANY(HAS_HOTEND, HAS_HEATED_BED, HAS_FAN)
    #define HAS_PREHEAT 1
  #endif
#endif
#if !HAS_PREHEAT
  #undef PREHEAT_SHORTCUT_MENU_ITEM
  #undef DGUS_PREHEAT_UI
#endif
`
HAS_MARLINUI_MENU was LCD before and DGUS_PREHEAT_UI works fine, but with new version if i enable 
DGUS_LCD_UI_MKS in Configuration.h, then in ..\lcd\extui\dgus\DGUSScreenHandler.cpp i cannot work with
preheat because from Configuration.h/adv is DGUS_PREHEAT_UI enabled, but with this in Conditionals_post.h
is disabled, with no reason.
Shouldn't be there
`#if HAS_TEMPERATURE && EITHER(EXTENSIBLE_UI, HAS_DWIN_E3V2)`?

### Bug Timeline

found it in new builds, last working build was from 10.11.2021

### Expected behavior

ENABLED(DGUS_PREHEAT_UI) really enabled

### Actual behavior

ENABLED(DGUS_PREHEAT_UI)  disabled

### Steps to Reproduce

1. download any new Marlin version
2. in configuration.h in line 2756 enable DGUS_LCD_UI_MKS
3. go see configuration_adv.h line 1787 and check if DGUS_PREHEAT_UI is enabled (should be)
4. go ..\lcd\extui\dgus\DGUSScreenHandler.cpp and check line 623 - #if ENABLED(DGUS_PREHEAT_UI)  (disabled)

### Version of Marlin Firmware

02010000

### Printer model

Fox MK1

### Electronics

does not matter

### Add-ons

does not matter

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Additional information & file uploads

None

ellensp, 2022-04-22:
replace #if HAS_TEMPERATURE && EITHER(HAS_MARLINUI_MENU, HAS_DWIN_E3V2)  In Conditionals_post.h
with #if HAS_TEMPERATURE && ANY(HAS_MARLINUI_MENU, HAS_DWIN_E3V2,HAS_DGUS_LCD_CLASSIC)

ellensp, 2022-04-22:
This PR fixes it.

closed 2022-04-22 by ellensp