MarlinFirmware/Marlin #24055: [BUG] Part Cooling fan turns on at boot after update to 2.0.9.3

lsellens, 2022-04-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Ever since flashing to 2.0.9.3 my part cooling fan turns on at boot. The LCD fan graphic does not indicate it is on. If I send a M107 it stays on. In order to turn it off I have to first turn it on, and then I can turn it off with M107. Flashed back to 2.0.9.2 to verify, and the bug is not present in that version. My board uses pin 4 as FAN_PIN. I do use a relay to control my power supply. I disabled that feature and removed the relay to no effect. Printer still turned on with part cooling fan running non stop.

### Bug Timeline

New to version 2.0.9.3

### Expected behavior

Part cooling fan should not be on

### Actual behavior

Part cooling fan starts on and wont turn back off until command is issued to turn it on and then off

### Version of Marlin Firmware

2.0.9.3

### Printer model

MKS Smelzi

### Host Software

OctoPrint

lsellens, 2022-04-18:
reverting #23181 fixed it

InsanityAutomation, 2022-04-18:
at a glance im wondering if it was meant to be initialized to 0 not 255.

thisiskeithb, 2022-04-18:
> at a glance im wondering if it was meant to be initialized to 0 not 255.

@descipher thought the same: https://github.com/MarlinFirmware/Marlin/pull/23181#pullrequestreview-853573579

descipher, 2022-04-18:
Yes it should be 0.

lsellens, 2022-04-18:
I tried changing it to 0 but that does not resolve my issue. Only changing `bool fans_need_update = true;` fixes it for me

descipher, 2022-04-19:
The correct init value should be 0, the issue here is that the planner will not update the fans unless the fans_need_update flag is set. So during the fan speed array init nothing changes since the planner does not see the flag as true.

```TERN_(HAS_TAIL_FAN_SPEED, if (fans_need_update) sync_fan_speeds(tail_fan_speed));```

The will only update if HAS_TAIL_FAN_SPEED is set so this is not a proper init.
  
The init should call -> ```planner.sync_fan_speeds(0);```and not mess with the flag because this is a safety issue when LASER_SYNCHRONOUS_M106_M107 is defined.

descipher, 2022-04-22:
Took a closer look at this.
The tail_fan_speed array should not be instantiated in the check_axes_activity() subroutine.
It kills performance to init in the stack pushing and popping the stack on every check_axes_activity call which is very frequent and actually takes more ram in most cases to run it this way. By moving it out as a global it can init the fans_need_update to true and properly init the fans on boot. It actually saves 4 bytes of ram by moving it out, however it does use more flash space. I would go with performance on this one. Also the array should be 128 to ensure the 1st call triggers the fans update test. The update code ignores a 0 or 255 value in order to only update on a fan value change.

descipher, 2022-04-23:
After testing, 0 is the correct initialization value .. lol

thisiskeithb, 2022-04-23:
Closing since there’s now a PR: https://github.com/MarlinFirmware/Marlin/pull/24076

closed 2022-04-23 by thisiskeithb

descipher, 2022-04-26:
Please confirm that you ran bugfix-2.0.x because your comments do not mention it.

lsellens, 2022-04-28:
> Please confirm that you ran bugfix-2.0.x because your comments do not mention it.

Are you referring to before or after your PR?

lsellens, 2022-04-29:
> ### Did you test the latest `bugfix-2.0.x` code?
> Yes, and the problem still exists.

descipher, 2022-04-29:
@lsellens Ok, Thanks.

lsellens, 2022-05-01:
@descipher So maybe I made a mistake. I could swear before creating this issue I flashed the bugfix code and still had the bug. I just did it again with the intention of adding your PR to it for further testing and the fan is not turning on and I have not added your changes yet. Maybe I was on the wrong branch or my flash failed. It was a week ago so I'm not certain what I did.

lsellens, 2022-05-01:
I don't think its relevant, but in case I missed something here is exactly what I just flashed to my printer, and my part cooling fan is not turning on at boot.
https://github.com/lsellens/Marlin/tree/MKS_SMELZI_V1.0-bugfix-2.0.x

descipher, 2022-05-01:
No worries, I was asking due to a different reason. Yes it should have worked with bugfix in your case but there is an option that can be set that would actually fail without the PR. It is not commonly used. I asked because I looked at your config and it was not enabled. Fan scaling if your curious. It sets the FAN to be a value of 128 by default and that's what the array was set to so it would leave it on in rare cases.