MarlinFirmware/Marlin #24058: [BUG] Backlash private variables used in ui_api

rondlh, 2022-04-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Here a section from ui_app.cpp, it uses backlash.distance_mm, backlash.correction and backlash.smoothing_mm.
```
  #if ENABLED(BACKLASH_GCODE)
    float getAxisBacklash_mm(const axis_t axis)       { return backlash.distance_mm[axis]; }
    void setAxisBacklash_mm(const_float_t value, const axis_t axis)
                                                      { backlash.distance_mm[axis] = constrain(value,0,5); }

    float getBacklashCorrection_percent()             { return ui8_to_percent(backlash.correction); }
    void setBacklashCorrection_percent(const_float_t value) { backlash.correction = map(constrain(value, 0, 100), 0, 100, 0, 255); }

    #ifdef BACKLASH_SMOOTHING_MM
      float getBacklashSmoothing_mm()                 { return backlash.smoothing_mm; }
      void setBacklashSmoothing_mm(const_float_t value) { backlash.smoothing_mm = constrain(value, 0, 999); }
    #endif
  #endif
```

But in backlash.h these 3 variables are set to private.

```
class Backlash {
public:
  static constexpr uint8_t all_on = 0xFF, all_off = 0x00;
    static uint8_t correction;
    static xyz_float_t distance_mm;
private:
  static axis_bits_t last_direction_bits;
  static xyz_long_t residual_error;

  #if ENABLED(BACKLASH_GCODE)
    static uint8_t correction;
    static xyz_float_t distance_mm;
    #ifdef BACKLASH_SMOOTHING_MM
      static float smoothing_mm;
    #endif
```

So enabling both BACKLASH_CORRECTION and EXTENSIBLE_UI will cause some compilation errors.

### Bug Timeline

Don't know

### Expected behavior

Marlin should be able to compile

### Actual behavior

Compilation errors. variables are private within context.

### Steps to Reproduce

1. Enable both BACKLASH_CORRECTION and EXTENSIBLE_UI
2. Compile Marlin

### Version of Marlin Firmware

Marlin 2.0.9.3 latest bugfix

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### Add-ons

BTT TFT35 V3.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Other (explain below)

borland1, 2022-04-22:
I noticed compile errors a few week ago, when enabling `BACKLASH_COMPENSATION` in `Configuration_adv.h` with a recent bugfix-2.0.x.   Something in file `backlash.cpp` error about "no member function declared in class 'Backlash'
```
void Backlash::set_correction_unt8(const uint8_t v) {
void Backlash::set_distance_mm(const AxisEnum axis, const float v) { 
```

I'm not enabling `BACKLASH_SMOOTHING` or `BACKLASH_GCODE`.  But I'm using the classic LCD menu UI, and noticed when doing manual mesh bed leveling, that the backlash compensation is not being applied while doing the **Manual Mesh Bed Leveling**.  This makes finding the sweet spot on each mesh point difficult and time consuming to correct.  Fortunately the workaround  to run print tests and adjust the mesh grid with Edit Mesh LCD menu item.

ellensp, 2022-04-23:
@borland1 fix that issue.  larger issue of attempting to use private variables not addressed yet

ellensp, 2022-04-23:
@rondlh these have already been fixed in bugfix https://github.com/MarlinFirmware/Marlin/commit/6b7868d943c
Part of https://github.com/MarlinFirmware/Marlin/pull/23826

closed 2022-04-23 by ellensp