MarlinFirmware/Marlin #24068: [BUG] (bug summary)

hs11701, 2022-04-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

when you uncomment eeprom settings while compiling marlin for SKR mini e3 v3 it successfully builds in VScode but when flashed to the main board causes a bootloop on the printer. this was found after a couple weeks going through each setting in configuration.h to find what was causeing it to be stuck on bootup and foun d that the uncomiting the eeprom setings option in marlin cause the boot loop tested on two diffrent SKR mini e3 V3 board with same results.

### Bug Timeline

unkown/ 4/1/22 when trying to boot new skr mini E3 V3

### Expected behavior

expected to boot up printer and run as normal

### Actual behavior

stuck in bootloop

### Steps to Reproduce

1. Compile marlin
2. uncomment #define EEPROM_SETTINGS
3. BUILD marlin
4. flash SKR MINI E3 V3 board 
5. stuck in bootup
6. repeat but leave EEprom Settings disabled and it successfully boots up on printer but unable to store any changes in settings.

### Version of Marlin Firmware

2.0.9.3

### Printer model

ender 3 pro

### Electronics

SKR MINI E3 V3

### Add-ons

BTT UPS, BTT relay, CR touch, Filament runout sensor, stock cr10 screen

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Additional information & file uploads

[eeprom active.zip](https://github.com/MarlinFirmware/Marlin/files/8541991/eeprom.active.zip)
[eeprom disabled working build confg.zip](https://github.com/MarlinFirmware/Marlin/files/8542002/eeprom.disabled.working.build.confg.zip)

Arakon, 2022-04-23:
I've used the Mini E3 v3 with 2.0.9.3 as well as multiple recent bugfix versions and eeprom active with no issue. Are you sure it's not a hardware issue, i.e. defective eeprom chip?

thisiskeithb, 2022-04-23:
> I've used the Mini E3 v3 with 2.0.9.3 as well as multiple recent bugfix versions and eeprom active with no issue. Are you sure it's not a hardware issue, i.e. defective eeprom chip?

Ditto. No issues here with an SKR Mini E3 V3 running 9188f57. This is a hardware issue.

closed 2022-04-23 by thisiskeithb