MarlinFirmware/Marlin #24070: SKR 2 with F429 compiles in PlatformIO, but doesn't upload the firmware to SD card

hcet14, 2022-04-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I don't know where to address this bug. PlatformIO, OpenOCD, or Marlin. I think I'm wrong here. Maybe someone can give me a hint and I'll close this issue imediatly!

Other people mentioned the same behaviour:
[https://drucktipps3d.de/forum/topic/firmware-marlin2-0-9-3-auf-btt-skr-2/?part=2](url)
[https://github.com/bigtreetech/SKR-2/issues/115](url)
[image]
Hit upload: compiles in PlatformIO, but doesn't upload the firmware to SD card

Marlin works. The firmware is build, but not uploaded to SD card.

Error message: 
Uploading .pio/build/BIGTREE_SKR_2_F429_USB/firmware.bin
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

embedded:startup.tcl:26: Error: Can't find board/stm32f429.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
*** [upload] Error 1

Environment: Visual Studio Code 1.66.2

### Bug Timeline

none, not important

### Expected behavior

Automatically upload to SD card after successful build in platformio.

### Actual behavior

No copying of firmware to SD card.

### Steps to Reproduce

See above

### Version of Marlin Firmware

doesn't matter

### Printer model

doesn't matter

### Electronics

doesn't matter

### Add-ons

doesn't matter

thisiskeithb, 2022-04-22:
You’ll need to copy firmware to the SD card manually and power cycle the board.

closed 2022-04-22 by thisiskeithb

hcet14, 2022-04-22:
> Of coarse. That's what I'm doing. I just wanted to know, where to address this little bug.

ellensp, 2022-04-23:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

hcet14, 2022-04-23:
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

Sorry, I apologize. Can you at least give me a hint where to address this bug? Thank you.

ellensp, 2022-04-23:
I answered you in discord

ellensp, 2022-04-23:
the only two ways a user can upload firmware on that controller  

1) you manually copy the firmware.bin onto a sdcard and insert the scard into the machine  and power it on.

2) you use a stlink hardware programmer. this is the default that platformio uses as it cannot do 1  (not advisable for the novice, 
you can wipe the bootloader) 

tThis is not a bug or an error.

Its is a lack of understanding on the part of the user.

hcet14, 2022-04-23:
Thank you I'm able to understand your answer. 

> Its is a lack of understanding on the part of the user.

Yes, you are absolutely right!

Thank you.

ellensp, 2022-04-23:
Have found a way that upload will launch a directory selector and ask you where to save it.   Is currently being tested

hcet14, 2022-06-17:
> I answered you in discord

@ellensp: Thanks a lot for the example. I'm not a programmer, as you can see, if you have a look into the code.
["https://github.com/bigtreetech/SKR-2/issues/161"](https://github.com/bigtreetech/SKR-2/issues/161)