MarlinFirmware/Marlin #24073: [BUG] (Skewed print with external driver settings in Marlin Bugfix2.0)

rouhit110, 2022-04-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

My printer cannot print long, vertical cubes perfectly flat and vertically. Print is skewed to X+ and Y+
I'm using a MKS Sbase V1.3 board that I have tried the bugfix2.0 firmware
Because I needed more power, I used Leadshine DM870 and DM556 external drivers
After many tests and incurring a lot of costs, I used an 8-bit MKS Gen1.4 board, which I saw in disbelief that the problem was solved.
After much research, I found out that the driver type selection settings in Marlin caused this to happen. I choose TB6600
I even chose TB6560 but the result was negative.
Can I solve my problem with another 32-bit board with a different brand? or is this problem solvable?

### Bug Timeline

I realized this problem after building and finishing my printer

### Version of Marlin Firmware

Bugfix2.0.x

### Printer model

Cartesian

### Add-ons

MKS PWC V3.0 , BLtouch , MKS tft70

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Additional information & file uploads

[image]
[image]
[Bugfix2.0 Configs.zip](https://github.com/MarlinFirmware/Marlin/files/8546806/Bugfix2.0.Configs.zip)

descipher, 2022-04-24:
DM556 spec:
[image]

Remark:
a) t1: ENA must be ahead of DIR by at least 5ms. Usually, ENA+ and ENA- are NC (not
connected). See “Connector P1 Configurations” for more information.
b) t2: DIR must be ahead of PUL active edge by 5ms to ensure correct direction;
c) t3: Pulse width not less than 2.5ms;
d) t4: Low level width not less than 2.5ms. 

Two things you need to consider:

The DM556 is slow 2.5ms even the old A4988 is 2.5us. 
The logic level minimum is 3.5v. A typical signal from a 32bit CPU is 3.3v  so this is likely the issue.
You could try pulling up the pulse pins to + 5v with a 10k resistor.

rouhit110, 2022-04-24:
> Two things you need to consider:
> 
> The DM556 is slow 2.5ms even the old A4988 is 2.5us. The logic level minimum is 3.5v. A typical signal from a 32bit CPU is 3.3v so this is likely the issue. You could try pulling up the pulse pins to + 5v with a 10k resistor.

Thank you very much for your guidance and explanation
Now it is connected to the pins + socket P1 of 3.3V voltage, however 10k resistor is needed?

thisiskeithb, 2022-04-24:
Also, how are you disabling the onboard drivers on the Sbase? Unless they are bypassed, you're running onboard + external drivers at the same time.

rouhit110, 2022-04-24:
> Also, how are you disabling the onboard drivers on the Sbase? Unless they are bypassed, you're running onboard + external drivers at the same time.

Yes, I use an onboard driver for the extruder
Should not it be so?

descipher, 2022-04-24:
You should be ok with the output connected to the existing 8825's. This issue with the MKS SBase 1.x board is it does not have a jumper to select a +3/5v drive on the outputs. So in order to validate the problem you need to do this with your Y output to confirm the issue is a voltage threshold.

[image]

If this corrects it then you have two choices. Switch the board out with an MKS SGEN_L V1.0 which has the +3/5v select jumper or add the pull ups to the externally driven axes.

thisiskeithb, 2022-04-24:
Closing since this isn’t a Marlin bug.

closed 2022-04-24 by thisiskeithb

rouhit110, 2022-04-24:
> If this corrects it then you have two choices. Switch the board out with an MKS SGEN_L V1.0 which has the +3/5v select jumper or add the pull ups to the externally driven axes.

Dear descipher, Thank you for your guidance and answering
But this jumper is on MKS Sbase board and I set it to 3.3v
However, should I do the wiring according to the schematic you provided for me?

thisiskeithb, 2022-04-24:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

descipher, 2022-04-24:
3.3v is wrong, you want the output to be 5v, I could not see any jumper on that board.

rouhit110, 2022-04-24:
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

Thanks dear thisiskeithb, 
But I did the same thing before and raised my problem on Facebook, but unfortunately I did not receive a response.

thisiskeithb, 2022-04-24:
Facebook isn’t the only option. Please use the other links / forums instead of this issue queue since this is not a Marlin bug.

rouhit110, 2022-05-14:
> Facebook isn’t the only option. Please use the other links / forums instead of this issue queue since this is not a Marlin bug.

Unfortunately, no one was able to guide me properly.
I was able to print well with an 8-bit controller and Marlin 1.1.7, but with the same controller and Marlin 1.1.9 my print was skewed.
I did the recommendations you see in this link but it did not work
Can you tell me what I should do ‌ Please?

ellensp, 2022-05-14:
This is still not a support forum.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).