MarlinFirmware/Marlin #24075: [BUG] (bug summary)

gnmrc-s, 2022-04-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Error while trying to home MP_SCARA

### Bug Timeline

old issue that people were able to solve with old marlin 2.0 variants

### Expected behavior

I expected it to home x than y and finally z

### Actual behavior

It homed only the z axis

### Version of Marlin Firmware

none

### Printer model

Custom SCARA machine

### Electronics

creality v 4.2.7 board

### Add-ons

none

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

ellensp, 2022-04-23:
please update subject and attach your config files.

ellensp, 2022-04-23:
also what does a M119 say when no endstops are being triggered?

gnmrc-s, 2022-04-23:
M119 command says open endstops

thisiskeithb, 2022-04-23:
Homing has been broken since 1.x: https://github.com/MarlinFirmware/Marlin/issues/10089

thisiskeithb, 2022-04-23:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/10089

closed 2022-04-23 by thisiskeithb

gnmrc-s, 2022-04-23:
I've read the other bug and it's not the same.
And also other people have made marlin 2.0 homing work.

gnmrc-s, 2022-04-23:
https://github.com/MarlinFirmware/Marlin/issues/20764

thisiskeithb, 2022-04-23:
> https://github.com/MarlinFirmware/Marlin/issues/10089

That's the long-standing "homing doesn't work on SCARA" thread, so let's use that.

gnmrc-s, 2022-04-25:
I've made a post about this problem on #10089, please when you have time go and check it out, it's 2 months tham I'm struggling trying to home my SCARA.