MarlinFirmware/Marlin #24078: EXTRUDER_AUTO_FAN_SPEED not working.[BUG] (bug summary)

Deema35, 2022-04-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

If I set parameter EXTRUDER_AUTO_FAN_SPEED to value not equal 255 my extruder fan not working. But if in pin file I change E0_AUTO_FAN_PIN to CONTROLLER_FAN_PIN, I can control fan even from Marlin menu.

### Bug Timeline

New

### Expected behavior

EXTRUDER_AUTO_FAN_SPEED must set fan speed.

### Actual behavior

Extruder fan not working if I set value not equal 255.

### Steps to Reproduce

1. Set EXTRUDER_AUTO_FAN_SPEED value not equal 255.

### Version of Marlin Firmware

2.0.9.3

### Electronics

BTT_SKR_MINI_V1_1

Bob-the-Kuhn, 2022-04-24:
Do you have **FAN_KICKSTART_TIME** enabled?  It applies to controller fans only.  Some fans require the kick start before they'll run at slower speeds.  I had to go to a PWM fan for my extruder because of this.

descipher, 2022-04-24:
Config files speak thousands of words ..

Deema35, 2022-04-24:
Yes I try FAN_KICKSTART_TIME and it dont have effect. And I thing this parameter affects only on fan witch blowing on model. This my config files

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/8550073/Config.zip)

thinkyhead, 2022-04-26:
I can't build `bugfix-2.0.x` using the provided configurations. The configs are for 2.0.9.3 and are missing values for `FIL_RUNOUT_PIN`, `CONTROLLER_FAN_PIN`, `X_CS_PIN` …

Deema35, 2022-04-26:
Sorry, this is my pin file:
[pins_BTT_SKR_MINI_V1_1.zip](https://github.com/MarlinFirmware/Marlin/files/8562724/pins_BTT_SKR_MINI_V1_1.zip)

descipher, 2022-04-26:
According the the board layout you only have one PWM connected fan available.

[image]

Deema35, 2022-04-26:
But I have Bed PWM pin where I connect second fan.

descipher, 2022-04-26:
Then try this:

pins_BTT_SKR_Mini_V1.1.h
```
#define HEATER_0_PIN                        PA8 // Extruder Heater
#define FAN_PIN                             PC8 // Part cooling
#define HEATER_BED_PIN                      -1 // not used 
#define FAN1_PIN                            PC9 //Extruder cooling fan
```

Configuration.h
```
#define E0_AUTO_FAN_PIN FAN1_PIN // E0 Auto fan enabled on FAN1_PIN
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed
```

Deema35, 2022-04-26:
To the HB pin  connected controller fan witch work good. I have trubles with E0_AUTO_FAN witch connected to FAN1 pin. Nevertheless if I set PC8 pin as controller fan pin I can controll fan, but if I try controll fan as E0_AUTO_FAN it dont woking.

descipher, 2022-04-26:
Did you try the posted changes?

descipher, 2022-04-26:
E.g. in the code it needs it!
```
#define INIT_E_AUTO_FAN_PIN(P) do{ if (P == FAN1_PIN || P == FAN2_PIN) { SET_PWM(P); SET_FAST_PWM_FREQ(P); } else SET_OUTPUT(P); }while(0) 
```

descipher, 2022-04-26:
You can switch PC8 and PC9 around but use the format given.
```
#define EXTRUDER_AUTO_FAN_SPEED 100  // to test pwm  
```

Deema35, 2022-04-26:
Thank you.This is work I set in pin.h: #define FAN1_PIN PC8
And In Configuration_adv.h: #define E0_AUTO_FAN_PIN FAN1_PIN

closed 2022-04-26 by Deema35