MarlinFirmware/Marlin #24085: [BUG] `ROTATE_PROGRESS_DISPLAY` errors out with Extruders 0

CAP1Sup, 2022-04-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When attempting to enable `ROTATE_PROGRESS_DISPLAY` on a machine without any extruders, the build will fail. I would have fixed this myself and proposed a PR, but I don't know enough about Marlin to understand how to properly fix the issue. The error is as follows: 
[image]

### Bug Timeline

Just found this today

### Version of Marlin Firmware

Latest bugfix-2.0.x

### Additional information & file uploads

[WITH_EXTRUDERS.zip](https://github.com/MarlinFirmware/Marlin/files/8556907/WITH_EXTRUDERS.zip)
[WITHOUT_EXTRUDERS.zip](https://github.com/MarlinFirmware/Marlin/files/8556908/WITHOUT_EXTRUDERS.zip) (causes error)

GMagician, 2022-04-25:
I think it should be replaced with a simple "E:". In this case text means "extimated" not extruder

thisiskeithb, 2022-04-25:
Closing since a PR has been created: https://github.com/MarlinFirmware/Marlin/pull/24087

PR #24087 (2022-04-26): Fix "elapsed" text
Elapsed text has been wrongly replaced by extruder text

Fix #24085
closed 2022-04-25 by thisiskeithb