MarlinFirmware/Marlin #24088: "Configuration Tool Under Construction" [BUG] (bug summary)

JeremyDWilliams, 2022-04-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

"Configuration Tool Under Construction" Shows for any option tried under the Configure option. Auto build was used. This is while using the autobuild Marlin Firmware Configuration Tool Extension with Visual Studio Code 1.66.2 and Marlin 2.0.X

### Bug Timeline

I don't know

### Expected behavior

Open a configuration panel

### Actual behavior

"Configuration Tool Under Construction" in red for all configure tabs.

### Steps to Reproduce

Click on Configure

### Version of Marlin Firmware

n/a

### Printer model

n/a

### Electronics

n/a

### Add-ons

n/a

### Additional information & file uploads

N/A

ellensp, 2022-04-26:
Its not implemented yet. Not a bug

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-04-26 by ellensp

JeremyDWilliams, 2022-04-29:
I apologize for not being more specific, frankly most interactions on github lean towards a (likely well earned) disrespect for users not 'doing what they are supposed to' instead of seeing their feedback as a potential source of ideas to improve code... but I digress :)

I perhaps wrongly assumed that this was possibly an issue that had come up before, and was just reappearing.

Here is why I submitted the bug report in a little more detail. The 'bug' is that it isn't necessarily a correct or accurate message. If a user doesn't bring in the right file name in the build, it displays the message as above. However, I respectively suggest it should point the user towards getting the missing files (or point to a site or file with more info). It isn't that the tool is under construction, it's that the user hasn't included it yet.

I have tested this. It comes up with the warning if it doesn't see the correct name in the build file. Replace it with the correct name(s) and it works fine. It isn't that the functionality hasn't been 'constructed', it just isn't seeing it. That's a less than subtle difference imo.

Is it a true bug? Maybe not in the sense of broken code, but maybe a user interface that is giving incorrect info.

And again, I apologize for not posting logs etc. as it's time consuming and I don't think is required. Just trying to help.

I do appreciate the user info - thank you. But I have no trouble using the software, I really did just want to help out your group a little :)

Thanks again, and congrats on working on a cool and useful project.

:)