MarlinFirmware/Marlin #24089: [FR] (Multiple firmwares)

aghriss, 2022-04-26:
### Describe the feature you want

I'm currently using Marlin 2.0.9.3 on SKR 1.4 Turbo. I realized that it's not possible to have it set for laser and use BLTouch as the same time. 

This made me wonder if it's possible to just have Marlin entry to switch between firmware. This is possible on SKR since it just requires re-naming the firmware file to `firmware.bin` and restarting the printer.
Similarly, it would also be nice to save EEPROM settings into sdcard (mesh, steps/per mm...) and load the ones we want depending on the firmware.

My C/C++ skills are above average, but it would take me a while to understand Marlin structure. If there is a guide I can follow and someone who wants to collaborate on this one, let me know.

thisiskeithb, 2022-04-26:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-04-26 by thisiskeithb

thinkyhead, 2022-04-26:
The firmware file must be flashed to the board before it can run, and then the board must reset. Firmware cannot simply be "loaded" from SD / USBFlash. A schema to move `firmware.bin` files from a folder to the root of the SD card would be the purview of something like the Firmware Flash plugin for OctoPrint.

aghriss, 2022-04-26:
> The firmware file must be flashed to the board before it can run, and then the board must reset. Firmware cannot simply be "loaded" from SD / USBFlash. A schema to move `firmware.bin` files from a [...]

I totally forgot about OctoPrint. Will try it. I guess that solves it!