MarlinFirmware/Marlin #24090: [BUG] WiiBoox One MightyBoard1280 Compilation error [Marlin-2.0.9.3]

shine7k, 2022-04-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin-2.0.9.3
Machine：Wiiboox One
Borad：MightyBoard
IDE：VSCODE

Build exception：error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board.
```
Processing MightyBoard1280 (board: megaatmega1280; platform: atmelavr@~3.4; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega1280.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Mega or Mega 2560 ATmega1280
HARDWARE: ATMEGA1280 16MHz, 8KB RAM, 124KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\MightyBoard1280\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\MightyBoard1280\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\MightyBoard1280\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\MightyBoard1280\src\src\HAL\AVR\Servo.cpp.o
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:50:0,
                 from Marlin\src\HAL\AVR\HAL.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:2290:4: error: #error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board."
   #error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:50:0,
                 from Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:2290:4: error: #error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board."
   #error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:50:0,
                 from Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:2290:4: error: #error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board."
   #error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:50:0,
                 from Marlin\src\HAL\AVR\Servo.cpp:55:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:2290:4: error: #error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board."
   #error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board."
    ^~~~~
*** [.pio\build\MightyBoard1280\src\src\HAL\AVR\Servo.cpp.o] Error 1
*** [.pio\build\MightyBoard1280\src\src\HAL\AVR\HAL.cpp.o] Error 1
*** [.pio\build\MightyBoard1280\src\src\HAL\AVR\MarlinSerial.cpp.o] Error 1
*** [.pio\build\MightyBoard1280\src\src\HAL\AVR\HAL_SPI.cpp.o] Error 1
```

**The configuration is as follows**

<platformio.ini>

```
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = MightyBoard1280
include_dir  = Marlin
............
[env:include_tree]
platform    = atmelavr
board       = megaatmega1280
build_flags = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__
src_filter  = +<src/MarlinCore.cpp>
```

<Configuration.h>
```
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_MIGHTYBOARD_REVE
#endif

```

### Steps to Reproduce

1. Modify profile：<platformio.ini><Configuration.h>
2. Build

### Version of Marlin Firmware

Marlin 2.0.9.3

### Printer model

WiibooxOne

ellensp, 2022-04-26:
This controller does not use standard Thermistors it uses a SPI MAX Thermocouple 

You need to select a thermocoupe based temperature sensor,   Ie type -2

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-04-26 by ellensp

shine7k, 2022-04-26:
> 该控制器不使用标准热敏电阻，它使用 SPI MAX 热电偶
> 
> 您需要选择基于热电偶的温度传感器，即 -2 型
> 
> 此问题队列用于处理 Marlin 错误报告和与开发相关的问题，我们不希望在这里处理用户支持问题。（如[本页](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question)所述。）为获得配置和故障排除帮助的最佳结果，请使用以下资源：
> 
> * MarlinFW.org [Marlin 文档](http://marlinfw.org)
> * RepRap.org[马林鱼论坛](http://forums.reprap.org/list.php?415)
> * Facebook 组[“马林固件”](https://www.facebook.com/groups/1049718498464482/)
> * Facebook 组[“用于 3D 打印机的 Marlin 固件”](https://www.facebook.com/groups/3Dtechtalk/)
> * [](https://www.youtube.com/results?search_query=marlin+configuration)YouTube 上的[马林鱼配置](https://www.youtube.com/results?search_query=marlin+configuration)
> * 3D 打印 Discord 服务器。加入链接：[https ://discord.gg/qNnSzsug](https://discord.gg/qNnSzsug)
> * Marlin 特定的 Discord 服务器。加入链接：[https ://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)
> 
> 在向社区寻求帮助后，如果共识指向 Marlin 中的错误，那么您应该发布[错误报告](https://github.com/MarlinFirmware/Marlin/issues/new/choose)。

> 

Thank you so much.