MarlinFirmware/Marlin #24091: [BUG] (bug summary)

gnmrc-s, 2022-04-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Bug described in #10089 
Please when you will have time help me, it’s two month that I’m struggling trying to fix this bug.

The bug is about MP SCARA homing and endstop not functioning.

### Bug Timeline

It’s an old issue, that someone was able to fix in old Marlin 2.0 versions

### Expected behavior

Explained in #24075 and #10089 in the last comment

### Actual behavior

Explained in #24075 and #10089 in the last comment

### Steps to Reproduce

Explained in #24075 and #10089 in the last comment

### Version of Marlin Firmware

None

### Printer model

Custom Serial Scara arm

### Electronics

Creality v4.2.7 board

### Add-ons

None

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Additional information & file uploads

Explained in #24075 and #10089 in the last comment

thisiskeithb, 2022-04-26:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/10089

thisiskeithb, 2022-04-26:
Please don’t open multiple issues for the same thing.

Since this project runs on volunteers, bugs may not be be worked on as quickly as you’d like, especially when it involves features/hardware not commonly used. If a developer has interest and the skills to fix the issue, they will.

You can also hop on our discord server and pay a developer directly if this is an immediate need.

closed 2022-04-26 by thisiskeithb

gnmrc-s, 2022-04-26:
Sorry, I know it’s wrong but I thought having an open issue in the front
page could make more people view my problem.

Keep up the great work, I’m gonna wait a response on #10089 when someone
will know how to fix it.