MarlinFirmware/Marlin #24094: [BUG] No consistency with ABL (bilinear)

Randwic, 2022-04-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello, this is the first PR that I post here, so sorry for the errors. I also use google translator

I have a problem with the ABL of my 2 FLSun SR, I can do 5 probes afterwards the results are never the same. I did the test with 2 probes (stock probes) and Euclid probe. I have the same problem with SKR2 and Nano Robin. Unprobed points are 0, except that I have unprobed points with a value.

**1st Probe :**
      0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +0.000 +0.033 +0.000 +0.000 +0.000 +0.000
 1 -0.016 +0.000 +0.080 +0.072 +0.072 +0.046 +0.000 +0.000 +0.000
 2 +0.000 +0.012 +0.047 +0.046 +0.046 +0.037 +0.013 +0.021 +0.000
 3 +0.000 -0.040 -0.007 -0.008 -0.001 +0.017 +0.007 +0.028 +0.000
 4 -0.081 -0.085 -0.074 -0.049 -0.018 +0.004 +0.018 +0.049 +0.071
 5 +0.000 -0.080 -0.059 -0.032 -0.029 -0.007 -0.007 +0.008 +0.000
 6 +0.000 -0.019 -0.071 -0.073 -0.056 -0.041 -0.017 -0.011 +0.000
 7 +0.000 +0.000 -0.043 -0.069 -0.079 -0.066 -0.092 +0.000 +0.000
 8 +0.000 +0.000 +0.000 +0.000 -0.085 +0.000 +0.000 +0.000 +0.000
[image]
 
**2nd probe :**
    0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +0.000 +0.047 +0.000 -0.034 +0.000 +0.000
 1 +0.000 +0.000 +0.077 +0.085 +0.071 +0.034 -0.012 +0.000 +0.000
 2 +0.000 +0.012 +0.034 +0.058 +0.059 +0.049 +0.038 +0.032 +0.000
 3 +0.000 -0.003 +0.018 +0.017 +0.024 +0.029 +0.007 +0.065 +0.000
 4 -0.043 -0.048 -0.049 -0.024 +0.007 +0.029 +0.043 +0.061 +0.084
 5 +0.000 -0.055 -0.034 -0.007 +0.009 +0.018 +0.005 +0.033 +0.000
 6 +0.005 +0.005 -0.021 -0.048 -0.044 -0.029 +0.008 -0.005 +0.000
 7 +4294967295.429496729542949672954294967295 +0.000 -0.018 -0.044 -0.042 -0.053 -0.067 +0.000 +0.000
 8 +0.000 +0.000 +0.000 +0.000 -0.060 +0.000 +0.000 +0.000 +0.000
[image]

**3rd probe :**  
  0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +0.000 +0.033 +0.000 +0.000 +0.000 +0.000
 1 +0.000 +0.000 +0.092 +0.072 +0.072 +0.046 +0.000 +0.000 +0.000
 2 +0.000 +0.024 +0.047 +0.058 +0.059 +0.037 +0.013 +0.021 +0.000
 3 +0.000 -0.028 -0.007 -0.008 -0.001 +0.017 +0.007 +0.028 +0.000
 4 -0.068 -0.073 -0.074 -0.049 -0.018 +0.004 +0.018 +0.049 +0.071
 5 +0.000 -0.075 -0.047 -0.027 -0.026 -0.007 -0.007 +0.008 +0.000
 6 +0.000 -0.019 -0.046 -0.061 -0.056 -0.042 -0.017 -0.018 +0.000
 7 +0.000 +0.000 -0.030 -0.069 -0.067 -0.066 -0.092 +0.000 +0.000
 8 +0.000 +0.000 +0.000 +0.000 -0.072 +0.000 +0.000 +0.000 +0.000
[image]

**4th probe :** 
      0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +0.000 +0.047 +0.000 +0.000 +0.000 +0.000
 1 +0.000 +0.000 +0.080 +0.072 +0.059 +0.034 +0.000 -0.007 +0.000
 2 +0.000 +0.012 +0.034 +0.046 +0.046 +0.037 +0.025 +0.008 +0.000
 3 +0.000 -0.040 -0.007 -0.008 -0.001 +0.017 +0.007 +0.040 +0.000
 4 -0.068 -0.073 -0.074 -0.049 -0.018 +0.017 +0.030 +0.049 +0.071
 5 +0.000 -0.067 -0.047 -0.019 -0.004 +0.006 +0.005 +0.020 +0.000
 6 +0.000 -0.020 -0.046 -0.061 -0.056 -0.017 +0.008 -0.018 +0.000
 7 +0.000 +0.000 -0.018 -0.044 -0.054 -0.053 -0.067 +0.000 +0.000
 8 +0.000 +0.000 +0.013 +0.000 -0.060 +0.000 +0.000 +0.000 +0.000
[image]

 **5th probe :** ( i have restart FLSUN)
  0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +0.000 +0.085 +0.000 +0.000 -0.023 +0.000
 1 +0.000 +0.000 +0.130 +0.122 +0.122 +0.096 +0.038 +0.000 +0.000
 2 +0.000 +0.062 +0.097 +0.108 +0.109 +0.087 +0.075 +0.058 +0.000
 3 +0.000 +0.022 +0.055 +0.054 +0.062 +0.076 +0.047 +0.102 +0.000
 4 +0.007 -0.010 +0.001 +0.013 +0.044 +0.067 +0.080 +0.099 +0.121
 5 +0.000 -0.005 +0.003 +0.031 +0.046 +0.043 +0.055 +0.070 +0.000
 6 +0.000 +0.044 +0.016 -0.002 +0.009 +0.021 +0.045 +0.045 +0.000
 7 +0.000 +0.000 +0.045 +0.006 -0.004 -0.003 -0.030 +0.000 +0.000
 8 +0.000 +0.000 +2.018 +0.000 -0.010 +0.000 +0.000 +0.000 +0.000
[image]

**6th probe :**  
 0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +0.000 +0.110 +0.000 +0.000 +0.000 +0.000
 1 +0.000 +0.000 +0.130 +0.122 +0.134 +0.096 +0.050 +0.000 +0.000
 2 +0.000 +0.074 +0.097 +0.108 +0.109 +0.099 +0.088 +0.071 +0.000
 3 +0.000 +0.022 +0.055 +0.054 +0.062 +0.076 +0.060 +0.115 +0.000
 4 -0.018 -0.023 -0.011 +0.013 +0.044 +0.067 +0.080 +0.099 +0.134
 5 +0.000 -0.017 +0.003 +0.031 +0.046 +0.043 +0.055 +0.070 +0.000
 6 +0.000 +0.031 +0.004 -0.011 +0.006 +0.009 +0.058 +0.045 +0.000
 7 +0.000 +0.000 +0.020 -0.007 -0.004 -0.003 -0.030 +0.000 +0.000
 8 +0.000 +0.000 +0.000 +0.000 -0.022 +4294967295.429496729542949672954294967295 +0.000 +0.000 +0.000
[image]

### Bug Timeline

New Bug

### Expected behavior

I seek to have a minimum of coherence of the Mesh Leveling

### Actual behavior

The abl is not constant

### Version of Marlin Firmware

Marlin 2.0.9.3 bugfix (Apr 26 2022 17:27:13) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Super Racer

### Printer model

FlSun SR

### Electronics

TFT 70

### Bed Leveling

UBL Bilinear mesh

Randwic, 2022-04-26:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:
> 
> 
> 
> - Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
> 
> - Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
> 
> - Connect to your printer from host software such as Cura, Printrun or Repetier Host.
> 
> - Send `M502` and `M500` to ensure your Configurations are applied.
> 
> - Issue the command `M111 S247` to enable maximum logging.
> 
> - Perform a `G28` to do your standard homing procedure.
> 
> - Do a `G29` to probe the bed. This will also enable bed leveling.
> 
> - Do some of the moves that revealed problems before. Take notes.
> 
> - Copy the log output into a **.TXT file** and attach it to your next reply.
> 
> 
> 
> Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

I will try that tomorrow

borland1, 2022-04-26:
Some inconsistancies with the probes is normal.  The Visualizer exaggerates the non-planar variations.  Your probe data appears show your bed to be quite flat.  How is that affecting the prints?  Have you tried printing a first layer bed leveling calibration pattern?

[image]

Randwic, 2022-04-26:
> Some inconsistancies with the probes is normal.  The Visualizer exaggerates the non-planar variations.  Your probe data appears show your bed to be quite flat.  How is that affecting the prints?  Have you tried printing a first layer bed leveling calibration pattern?
> 
> 
> 
> [image]
> 
> 

hello, I understand your point, I have not tried to print a test.  what I don't understand is that I have huge inconsistencies between 2 probes.  you can see in the photos that I sometimes have 8 digit probes.  Also, since the last bug fix, unprobed points should be 0 but this is not always the case.  I hope you will understand with the translation

Bob-the-Kuhn, 2022-04-26:
I see that all the strange probe results are on the edges of the bed.  Maybe some of the probes are just missing the edge of the bed.  Try increasing **PROBING_MARGIN** in configuration.h to see if that helps.

Randwic, 2022-04-27:
> Je vois que tous les résultats des sondes étranges sont sur les bords du lit. Peut-être que certaines des sondes manquent juste le bord du lit. Essayez d'augmenter **PROBING_MARGIN** dans configuration.h pour voir si cela aide.

I checked, the probe does not magnify any point. i have this in configuration.h : #define PROBING_MARGIN 5

Randwic, 2022-04-27:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:
>
> * Download Marlin [...]

How to have access to the whole log, because with pronterface, I cannot copy the whole log. the lines fade away

Randwic, 2022-04-27:
The problem appeared after these fixes

#23868
#24009
#24002

Guilouz, 2022-04-27:
Issue appears on latest bugfix since bilinear leveling refactor with this PR #23868. 

Sometimes certain points are not recorded in the mesh and have no value (and the printer freezes when it prints and its displacement falls on a non-existent point in the mesh), sometimes certain points have an extreme value of the type +4294967295.429496729542949672954294967295, sometimes some unprobed points are not all 0.0.

Tested with and without :
#define EXTRAPOLATE_BEYOND_GRID
#define ABL_BILINEAR_SUBDIVISION

[Marlin Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8570592/Marlin.Configuration.zip)

Randwic, 2022-04-29:
@tombrazier, Do you have an idea for this problem? Thanks in advance

tombrazier, 2022-04-29:
> Sometimes certain points are not recorded in the mesh and have no value ... sometimes certain points have an extreme value of the type +4294967295.429496729542949672954294967295, sometimes some unprobed points are not all 0.0.

@Guilouz +4294967295.429496729542949672954294967295 is obviously a problem. Could you specifically point out where the other two issues are occurring?

tombrazier, 2022-04-29:
Can someone give #24107 a go please.

Randwic, 2022-04-29:
> Quelqu'un peut-il [essayer # 24107](https://github.com/MarlinFirmware/Marlin/pull/24107) s'il vous plaît.

I will try that this afternoon

tombrazier, 2022-04-29:
Merci.

Randwic, 2022-04-29:
You are french?

Randwic, 2022-04-29:
I applied the modification, but the results are the same. The first probe is ok then nok

PROBE 1

0      1      2      3      4      5      6      7      8
 0 +0.086 +0.065 +0.050 +0.045 +0.027 +0.008 -0.034 -0.074 -0.052
 1 +0.015 +0.050 +0.044 +0.054 +0.047 +0.021 -0.014 -0.000 +0.044
 2 -0.085 -0.001 +0.038 +0.050 +0.074 +0.070 +0.032 +0.046 +0.099
 3 -0.135 -0.054 -0.026 +0.011 +0.022 +0.047 +0.035 +0.062 +0.136
 4 -0.099 -0.089 -0.065 -0.039 -0.004 +0.023 +0.062 +0.079 +0.096
 5 -0.103 -0.073 -0.052 -0.002 +0.011 +0.033 +0.030 +0.031 +0.070
 6 -0.058 -0.008 -0.029 -0.036 -0.029 +0.008 +0.029 +0.021 +0.018
 7 +0.012 +0.018 +0.024 -0.004 -0.004 -0.003 -0.019 +0.001 -0.000
 8 +0.081 +0.066 +0.034 -0.002 -0.009 -0.001 -0.012 -0.030 -0.031
 
 PROBE 2
 
   0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +0.000 +0.027 +0.000 +0.000 +0.000 +0.000
 1 +0.000 +0.000 +0.044 +0.042 +0.034 +0.021 -0.014 +0.000 +0.000
 2 +0.000 -0.014 +0.025 +0.050 +0.062 +0.070 +0.044 +0.034 +0.000
 3 +4294967295.429496729542949672954294967295 -0.054 -0.026 -0.007 +0.034 +0.035 +0.035 +0.062 +0.000
 4 -0.112 -0.089 -0.077 -0.039 +0.008 +0.035 +0.062 +0.079 +0.109
 5 +0.000 -0.061 -0.040 -0.014 +0.011 +0.022 +0.036 +0.031 +0.000
 6 +0.000 -0.014 -0.029 -0.027 -0.016 -0.004 +0.017 +0.008 +0.000
 7 +0.000 +0.000 +0.024 -0.004 -0.016 -0.003 -0.031 +0.000 +0.000
 8 +0.000 +0.000 +0.000 +0.000 -0.009 +0.000 +0.000 +0.000 +0.000
 
 PROBE 3
 
0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +4294967295.429496729542949672954294967295 +0.000 +0.014 +0.000 +0.000 +0.000 +0.000
 1 +0.000 +0.000 +0.044 +0.042 +0.047 +0.008 -0.026 +0.000 +0.000
 2 +0.000 -0.013 +0.025 +0.037 +0.037 +0.058 +0.019 +0.021 +0.000
 3 +0.000 -0.054 -0.030 -0.007 +0.009 +0.035 +0.023 +0.037 +0.000
 4 -0.087 -0.089 -0.065 -0.039 +0.008 +0.035 +0.062 +0.079 +0.109
 5 +0.000 -0.048 -0.027 +0.011 +0.011 +0.033 +0.030 +0.031 +0.000
 6 +0.035 +0.017 -0.016 -0.011 -0.004 +0.033 +0.050 +0.032 +0.000
 7 +0.000 +4294967295.429496729542949672954294967295 +0.036 +0.021 -0.004 +0.022 -0.006 +0.000 +0.000
 8 +0.000 +0.000 +0.044 +0.000 +0.003 +4294967295.429496729542949672954294967295 +0.000 +0.000 +0.000

Randwic, 2022-04-29:
the following

 **PROBE 4 G29 with Pronterface without G28**
 
  0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +0.000 +0.027 +0.000 +0.000 +0.000 +0.000
 1 +0.000 +0.000 +0.032 +0.029 +0.034 +0.021 -0.014 +0.000 +0.000
 2 +0.000 -0.014 +0.025 +0.037 +0.037 +0.058 +0.032 +0.022 +0.000
 3 +0.000 -0.054 -0.043 -0.007 +0.009 +0.035 +0.035 +0.049 +0.000
 4 -0.099 -0.089 -0.065 -0.039 -0.004 +0.035 +0.062 +0.092 +0.109
 5 +0.000 -0.061 -0.052 -0.014 +0.024 +0.022 +0.036 +0.049 +0.000
 6 +0.000 -0.014 -0.029 -0.027 -0.029 +0.008 +0.025 +0.007 +0.000
 7 +0.000 +0.000 +0.024 +0.009 -0.005 -0.006 -0.006 +0.000 +0.000
 8 +0.000 +0.000 +0.000 +0.000 +0.003 +0.000 +0.000 +0.000 +0.000
 
 **PROBE 5 G29 with Pronterface without G28**
 
   0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +0.000 +0.014 +0.000 -0.027 +0.000 +4294967295.429496729542949672954294967295
 1 +0.000 +0.000 +0.032 +0.042 +0.047 +0.021 -0.014 +0.004 +0.000
 2 +0.000 -0.014 +0.025 +0.037 +0.049 +0.058 +0.032 +0.046 +0.000
 3 +0.000 -0.054 -0.018 -0.007 +0.034 +0.035 +0.035 +0.062 +0.000
 4 -0.099 -0.089 -0.065 -0.039 +0.008 +0.048 +0.075 +0.092 +0.134
 5 +0.000 -0.061 -0.040 -0.002 +0.036 +0.045 +0.055 +0.056 +0.000
 6 +0.000 +0.005 -0.004 -0.011 +0.009 +0.046 +0.067 +0.058 +0.000
 7 +0.000 +0.000 +0.049 +0.021 +0.008 +0.019 +0.019 +0.000 +0.000
 8 +0.000 +0.000 +0.000 +0.000 +0.028 +0.000 +4294967295.429496729542949672954294967295 +0.000 +0.000
 
 **PROBE 6 G29 with Pronterface without G28**
 
  0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +0.000 +0.014 +0.000 +0.000 +0.000 +0.000
 1 +0.000 +0.000 +0.044 +0.042 +0.034 +0.008 -0.026 +0.000 +0.000
 2 +0.000 -0.026 +0.025 +0.025 +0.049 +0.058 +0.032 +0.022 +0.000
 3 +0.000 -0.067 -0.028 -0.014 +0.009 +0.035 +0.023 +0.049 +0.000
 4 -0.112 -0.101 -0.077 -0.051 -0.004 +0.023 +0.062 +0.079 +0.099
 5 +0.000 -0.073 -0.052 -0.002 +0.011 +0.008 +0.043 +0.056 +0.000
 6 +0.000 -0.008 -0.041 -0.024 -0.029 +0.008 +0.042 +0.033 +4294967295.429496729542949672954294967295
 7 +0.000 +0.000 +0.024 -0.004 -0.005 -0.006 -0.006 +0.027 +0.000
 8 +0.000 +0.000 +0.000 +4294967295.429496729542949672954294967295 +0.003 +0.000 +0.000 +0.000 +0.000
 
 **PROBE 7 G28/G29 with Pronterface**
 
    0      1      2      3      4      5      6      7      8
 0 +0.000 +0.000 +0.000 +2.018 +0.027 +0.000 +0.000 +0.000 -0.011
 1 +2.018 +0.000 +0.044 +0.042 +0.047 +0.021 -0.014 +0.000 +0.000
 2 +0.000 -0.013 +0.013 +0.037 +0.049 +0.058 +0.032 +0.009 +0.000
 3 +0.000 -0.054 -0.005 -0.007 +0.017 +0.022 +0.010 +0.037 +0.000
 4 -0.087 -0.076 -0.065 -0.039 +0.008 +0.035 +0.062 +0.054 +0.096
 5 +0.000 -0.061 -0.040 +0.011 +0.024 +0.009 +0.024 +0.049 +0.000
 6 +0.000 +0.005 -0.016 -0.011 -0.029 +0.008 +0.017 +0.033 +0.000
 7 +0.000 +0.000 +0.036 -0.004 -0.005 +0.007 -0.019 +0.000 +0.000
 8 +0.000 +0.000 +0.000 +0.000 +0.003 +0.000 +0.000 +0.000 +0.000

tombrazier, 2022-04-29:
> You are french?

No. Just responding to your quote which translated my question into french. At least I think that was it.

I think you are missing my second commit. The code delta should look like [this](https://github.com/MarlinFirmware/Marlin/pull/24107/files#diff-68affdc6c35624021c07a639e1e2f7977115b927c88612facdb0d3991025bd49).

Randwic, 2022-04-29:
sorry @tombrazier , I'm a beginner. The file I need to copy is the Filesis in FIles Changed

Randwic, 2022-04-29:
Like this ?

#if ENABLED(AUTO_BED_LEVELING_BILINEAR)
      if (!abl.dryrun
        && (abl.gridSpacing != bbl.get_grid_spacing() || abl.probe_position_lf != bbl.get_grid_start())
     ) {
        // Reset grid to 0.0 or "not probed". (Also disables ABL)
        reset_bed_level();
        COPY(abl.z_values, Z_VALUES_ARR);

        // Can't re-enable (on error) until the new grid is written
        abl.reenable = false;
      }

      TERN_(IS_KINEMATIC, COPY(abl.z_values, Z_VALUES_ARR));
    #endif // AUTO_BED_LEVELING_BILINEAR

  } // !g29_in_progress

Randwic, 2022-04-29:
Thank you very much @tombrazier , the results are now super constant with the last commit. It's just amazing compared to what I had.

Could you explain the code you changed?

> PROBE 1

  0      1      2      3      4      5      6      7      8
 0 +0.073 +0.083 +0.080 +0.056 +0.026 +0.012 -0.025 -0.078 -0.091
 1 +0.014 +0.047 +0.055 +0.062 +0.049 +0.024 -0.023 -0.024 +0.009
 2 -0.047 +0.010 +0.035 +0.047 +0.048 +0.059 +0.027 +0.019 +0.085
 3 -0.097 -0.017 +0.019 +0.016 +0.037 +0.042 +0.019 +0.053 +0.137
 4 -0.060 -0.065 -0.040 -0.014 +0.006 +0.042 +0.030 +0.071 +0.106
 5 -0.085 -0.048 -0.013 +0.014 +0.026 +0.025 +0.031 +0.042 +0.084
 6 -0.043 +0.011 -0.013 -0.014 -0.016 -0.004 +0.021 +0.009 +0.037
 7 +0.022 +0.043 +0.052 +0.005 +0.008 -0.003 -0.015 -0.011 -0.010
 8 +0.107 +0.097 +0.058 +0.017 +0.003 +0.006 -0.011 -0.029 -0.050
 
 PROBE 2
 
  0      1      2      3      4      5      6      7      8
 0 +0.073 +0.083 +0.080 +0.056 +0.013 +0.012 -0.025 -0.078 -0.091
 1 +0.014 +0.047 +0.055 +0.050 +0.049 +0.011 -0.010 -0.024 +0.009
 2 -0.047 +0.010 +0.035 +0.047 +0.048 +0.059 +0.014 +0.019 +0.085
 3 -0.097 -0.017 +0.020 +0.006 +0.017 +0.030 +0.019 +0.053 +0.137
 4 -0.060 -0.065 -0.040 -0.026 +0.006 +0.029 +0.042 +0.058 +0.081
 5 -0.085 -0.048 -0.038 +0.014 +0.014 +0.025 +0.031 +0.042 +0.084
 6 -0.043 +0.023 -0.014 -0.014 -0.016 -0.004 +0.009 -0.004 +0.037
 7 +0.022 +0.043 +0.039 +0.005 -0.004 -0.003 -0.028 -0.011 -0.010
 8 +0.107 +0.097 +0.058 +0.017 +0.003 +0.006 -0.011 -0.029 -0.050  
 
 PROBE 3
 
  0      1      2      3      4      5      6      7      8
 0 +0.073 +0.083 +0.080 +0.056 +0.013 +0.012 -0.025 -0.078 -0.091
 1 +0.014 +0.047 +0.055 +0.050 +0.049 +0.011 -0.010 -0.024 +0.009
 2 -0.047 +0.010 +0.035 +0.047 +0.060 +0.059 +0.027 +0.019 +0.085
 3 -0.097 -0.017 +0.019 +0.016 +0.017 +0.042 +0.019 +0.053 +0.137
 4 -0.060 -0.065 -0.040 -0.026 +0.006 +0.029 +0.042 +0.071 +0.106
 5 -0.085 -0.048 -0.038 +0.014 +0.026 +0.025 +0.031 +0.042 +0.084
 6 -0.043 +0.011 -0.013 -0.014 -0.016 +0.009 +0.021 +0.009 +0.037
 7 +0.022 +0.043 +0.031 +0.005 +0.008 -0.003 -0.015 -0.011 -0.010
 8 +0.107 +0.097 +0.058 +0.017 +0.003 +0.006 -0.011 -0.029 -0.050
 
 PROBE 4 
 
   0      1      2      3      4      5      6      7      8
 0 +0.073 +0.083 +0.080 +0.056 +0.013 +0.012 -0.025 -0.078 -0.091
 1 +0.014 +0.047 +0.055 +0.050 +0.036 +0.011 -0.010 -0.024 +0.009
 2 -0.047 +0.010 +0.035 +0.047 +0.048 +0.047 +0.014 +0.019 +0.085
 3 -0.097 -0.017 +0.020 +0.018 +0.017 +0.030 +0.019 +0.040 +0.137
 4 -0.073 -0.077 -0.053 -0.026 -0.007 +0.017 +0.030 +0.058 +0.081
 5 -0.085 -0.048 -0.026 +0.001 +0.014 +0.012 +0.019 +0.029 +0.084
 6 -0.043 +0.011 -0.026 -0.014 -0.016 +0.000 +0.008 +0.021 +0.037
 7 +0.022 +0.043 +0.031 +0.005 -0.004 -0.003 -0.015 -0.011 -0.010
 8 +0.107 +0.097 +0.058 +0.017 +0.003 +0.006 -0.011 -0.029 -0.050
 
 PROBE 5 
 
 0      1      2      3      4      5      6      7      8
 0 +0.073 +0.083 +0.080 +0.056 +0.013 +0.012 -0.025 -0.078 -0.091
 1 +0.014 +0.047 +0.055 +0.050 +0.049 +0.011 -0.010 -0.024 +0.009
 2 -0.047 +0.010 +0.035 +0.047 +0.048 +0.038 +0.027 +0.019 +0.085
 3 -0.097 -0.017 +0.019 +0.004 +0.012 +0.027 +0.010 +0.051 +0.137
 4 -0.073 -0.065 -0.053 -0.026 +0.006 +0.029 +0.042 +0.071 +0.106
 5 -0.085 -0.048 -0.038 +0.014 +0.014 +0.025 +0.031 +0.042 +0.084
 6 -0.043 +0.011 -0.013 -0.014 -0.016 +0.009 +0.021 +0.009 +0.037
 7 +0.022 +0.043 +0.031 +0.005 -0.004 -0.003 -0.015 -0.011 -0.010
 8 +0.107 +0.097 +0.058 +0.017 -0.010 +0.006 -0.011 -0.029 -0.050  
 
 PROBE 6 
 
  0      1      2      3      4      5      6      7      8
 0 +0.073 +0.083 +0.080 +0.056 +0.013 +0.012 -0.025 -0.078 -0.091
 1 +0.014 +0.047 +0.055 +0.050 +0.049 +0.024 -0.010 -0.024 +0.009
 2 -0.047 -0.001 +0.035 +0.034 +0.048 +0.047 +0.002 +0.007 +0.085
 3 -0.097 -0.030 +0.008 +0.006 +0.030 +0.017 +0.006 +0.028 +0.137
 4 -0.073 -0.065 -0.053 -0.039 -0.007 +0.017 +0.030 +0.046 +0.081
 5 -0.085 -0.048 -0.026 +0.001 +0.014 +0.012 +0.019 +0.029 +0.084
 6 -0.043 -0.002 -0.013 -0.027 -0.028 -0.012 -0.004 -0.004 +0.037
 7 +0.022 +0.043 +0.031 -0.007 -0.004 -0.015 -0.028 -0.011 -0.010
 8 +0.107 +0.097 +0.058 +0.017 -0.010 +0.006 -0.011 -0.029 -0.050  
 
 PROBE 7 
 
 0      1      2      3      4      5      6      7      8
 0 +0.073 +0.083 +0.080 +0.056 +0.013 +0.012 -0.025 -0.078 -0.091
 1 +0.014 +0.047 +0.055 +0.050 +0.049 +0.024 -0.023 -0.024 +0.009
 2 -0.047 -0.002 +0.035 +0.034 +0.048 +0.047 +0.002 +0.007 +0.085
 3 -0.097 -0.030 +0.006 +0.004 +0.012 +0.030 +0.008 +0.053 +0.137
 4 -0.073 -0.077 -0.053 -0.039 +0.006 +0.029 +0.042 +0.071 +0.106
 5 -0.085 -0.061 -0.051 +0.001 +0.014 +0.012 +0.037 +0.042 +0.084
 6 -0.043 -0.002 -0.026 -0.027 -0.028 +0.000 +0.033 +0.021 +0.037
 7 +0.022 +0.043 +0.031 +0.005 -0.004 +0.010 -0.015 -0.011 -0.010
 8 +0.107 +0.097 +0.058 +0.017 +0.003 +0.006 -0.011 -0.029 -0.050

Randwic, 2022-04-29:
> PS if you know how to use a patch file (for example with `git apply`) then you can get a patch for any pull request by adding `.diff` to the end of the URL. So in this case you get a patch with [this URL](https://patch-diff.githubusercontent.com/raw/MarlinFirmware/Marlin/pull/24107.diff).

Thanks, but it's not for me ^^

tombrazier, 2022-04-29:
> Regarding the line (COPY(abl.z_values, Z_VALUES_ARR);) I no longer have it in my file. just this...

Yes, that's right.

thisiskeithb, 2022-04-29:
#24107 has been merged.

closed 2022-04-29 by thisiskeithb