MarlinFirmware/Marlin #24097: [BUG] FYSETC_MINI_12864_2_1 Backlight Won't Turn On

dckostin, 2022-04-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The FYSETC_MINI_12864 v2.1 display backlight will not turn on when I use the 'bugfix-2.0.x' branch of the code (needed to address my print fan issue), all other functionality working correctly.  The display functions properly using the '2.0.x' main branch.  If I download the '2.0.x' branch and then download the 'bugfix-2.0.x' branch, the backlight will remain illuminated at whatever color I last had, not the default color, and I am unable to change the display color.  If I then cycle power to the printer, the display will go dark until I reload the '2.0.x' branch.

Uncommenting the line "#define LED_PIN   PB14" in file 'pins_FYSETC_S6.h' does not resolve the issue.

### Bug Timeline

Noticed it immediately upon loading 'bugfix-2.0.x'

### Expected behavior

Expected the display to be illuminated and be able to change the display color from the Lights menu.

### Actual behavior

Display backlight does not illuminate so display is dark and is unresponsive to the Lights menu.

### Steps to Reproduce

1. Compile and download for a:
(line 99) #define MOTHERBOARD BOARD_FYSETC_SPIDER_V2_2
(line 110) #define SERIAL_PORT -1
(line 2601) #define FYSETC_MINI_12864_2_1      // Type A/B. NeoPixel RGB Backlight

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Voron 2.2

### Electronics

Fysetc Spider v2.2, Fysetc Mini 12864 v2.1

### Add-ons

n/a

### Additional information & file uploads

Note the variant.h file is from buildroot\share\PlatformIO\variants\MARLIN_FYSETC_S6\variant.
[Fysetc_Mini12864_Display_bug.zip](https://github.com/MarlinFirmware/Marlin/files/8578613/Fysetc_Mini12864_Display_bug.zip)
h

ellensp, 2022-04-28:
You have #define RGB_LED
But the FYSETC_MINI_12864_2_1 is a NEOPIXEL_LED not a RGB_LED

dckostin, 2022-04-28:
@ellensp, Thank you for pointing out my configuration error!  I had enabled RGB_LED after reading a complier error received from enabling LED_CONTROL_MENU ("Lights") to control the display color.  A more complete reading of the compiler error pointed me to enabling NEOPIXEL_LED instead.  Confusing why this worked in 2.0.x branch.

My apologies for incorrectly reporting this as a bug.

closed 2022-04-28 by ellensp