MarlinFirmware/Marlin #24104: [BUG] (BTT SKR GTR TMC CONNECTION ERROR)

Enzo-Enz, 2022-04-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi
there is this error on LCD when my printer turns on and stepper motor vibrating except of moving!

[image]

[image]

### Bug Timeline

old

### Version of Marlin Firmware

2.0.9.3

### Printer model

DIY printer

### Electronics

btt skr gtr/tmc2130

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/8589600/config.zip)

ellensp, 2022-04-29:
You need to have external power connected, not just usb power.

And you need to look at results of M122, is it all or just one?

You have 4 tmcs configured but only one is pictured plugged in, this would also cause this error. 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Enzo-Enz, 2022-04-29:
> You need to have external power connected, not just usb power.
>
> And you need to look at results of M122, is it all or just one?
>
> You have 4 tmcs configured but only one is pictured plugged in, this [...]

external power for motors is connected and I test that with 4 stepper drive which I enabled in configuration but the error is still exist!

ellensp, 2022-04-29:
still need to see M122 results

Enzo-Enz, 2022-04-29:
> still need to see M122 results

[M122.txt](https://github.com/MarlinFirmware/Marlin/files/8590340/M122.txt)

ellensp, 2022-04-29:
Check your stepper modules are actually configured for SPI mode.

[image]

The SPI pinsmust be empty and CFG4 and CFG5 should have the pins connected as pictured in Pink.

Enzo-Enz, 2022-04-29:
> Check your stepper modules are actually configured for SPI mode.
> 
> [image]
> 
> The SPI pads must be empty and CFG4 and CFG5 pads should be connected as pictured in Purple/Pink.

I think they have been set on STEP/DIR mode!

[image]

ellensp, 2022-04-29:
They have.
Not a marlin issue.

closed 2022-04-29 by ellensp