MarlinFirmware/Marlin #24112: [BUG] (MKS Sgen_L V1.0 not connecting to Controller)

Zatarines, 2022-04-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Latest Marlin bugfix 2.0.x, is not working with  REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER configured on a MKS Sgen_L V1.0. The FW compiles successfully w/ VScode, and I have uploaded several different configs. versions to the board and nothing seems to work. (only get a light up screen w/ all characters solid. squared like.) I tried working w/ a REPRAP_DISCOUNT_SMART_CONTROLLER that I have fully working on a CNC, but it looks the same. At this point I even tried a MKS TFT32_L v4.0, but not even that seems to work. I did look around on the ISSUES reported but nothing seems to resolve it. (The Flat Cables are Flipped and so the screens turns on.) I compared the PINs_ with a previous version of Marlin 2.0.x that had everything working fine on this board but I cant seem to find any difference. ANyone got a light on this issue? 

### Bug Timeline

NOn resolved issue

### Expected behavior

Any controller board to work with this set up

### Actual behavior

NO controller board is displaying image. 

### Steps to Reproduce

1; config MKS Sgen_L v1.0 to latest Marlin Bugfix w/ controller REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and have it to display. 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8595207/Marlin.zip)

### Version of Marlin Firmware

Marlin bugfix-2.0.x Apr 25, 2022

### Printer model

DYI Cartesian set up 

### Electronics

MKS Sgen_L v1.0

thisiskeithb, 2022-04-30:
Using your configs, I just flashed an MKS SGen_L V1.0 with a RepRap Discount Full Graphic LCD and it works fine.

I suspect a hardware/cable issue, especially with other peripherals not working.

closed 2022-04-30 by thisiskeithb

Zatarines, 2022-04-30:
I bought the same board to test the code. I appreciate your time. thank you @thisiskeithb