MarlinFirmware/Marlin #24115: [BUG] STOP button does not stop/reset the print status

joao-a-cardoso, 2022-04-30:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Pressing the Stop button stops the printing, but it's impossible to pofrom some operations, for example print the same model since the progress/status does not reset.

### Expected behavior

It should reset to state "ready", allowing all operations.

### Actual behavior

It does not reset the print status. It behaves presents the same menu options as it if where still printing.

The only workaround to print the same model from the beginning is that rebooting the printer by shutting it down.

### Steps to Reproduce

1. Start printing some file
2. After it starts, wait a while and select "Stop"
3. Check the menus, it presents the same options as it is still printing.

### Version of Marlin Firmware

Using a Maker Base MKS Robin Nano 1.3, with a TST35 display.

ellensp, 2022-05-01:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

joao-a-cardoso, 2022-05-04:
Hello, I already did that (see above, please). The wrong behaviour remains

closed 2022-08-08 by github-actions[bot]