MarlinFirmware/Marlin #24119: I can't get Marlin to compile

Newbmast3r69, 2022-05-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I can't compile the marlin firmware. I'm on the bugfix 2.0 branch and I have the config files for the Ender 3 pasted in. It goes fine for little bit, but then I am greeted by ugly yellow and red lines, which eventually leads to a Failure.

### Bug Timeline

5/1/2022

### Expected behavior

I expected it to compile fine and give me access to the bin file.

### Actual behavior

I tried to compile it. There was a bunch of errors and I got the resulting...

Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS C:\Users\26xie\Downloads\Marlin-bugfix-2.0.x> platformio run -e STM32F103RE_creality
Processing STM32F103RE_creality (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------Platform Manager: Installing ststm32 @ ~12.1
Unpacking  [####################################]  100%
Platform Manager: ststm32 @ 12.1.1 has been installed!
The platform 'ststm32@~12.1' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.5.2
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\STM32F103RE_creality\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\VirtIOSerial.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\WMath.cpp.o
[... 438 lines cut ...]

Ender 3

### Electronics

Creality 4.2.2 board

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

I used all default configuration files for both the normal and adv.h. Everything is stock except the Ender 3 profile.

Attached is a picture of the errors i was getting
[image]

thisiskeithb, 2022-05-01:
Looks like something broke with changes from https://github.com/MarlinFirmware/Marlin/pull/24106

thisiskeithb, 2022-05-01:
The changes from https://github.com/MarlinFirmware/Marlin/pull/24106 / https://github.com/MarlinFirmware/Marlin/commit/b37d13af7278c241064b61279e7e8538a03206a3 now break on any config that doesn't explicitly define driver types.

Before, A4988's were assumed when undefined.

Newbmast3r69, 2022-05-01:
So can I fix it or should I do something else like grab a stable version or wait for the Bugfix to be updated?

thisiskeithb, 2022-05-01:
Just uncomment X, Y, Z, and E driver types and it'll compile.

This will need to be addressed at a higher-level.

Newbmast3r69, 2022-05-01:
Okay thanks a lot! Your awesome and have a great day!

Newbmast3r69, 2022-05-01:
I tried that and it worked. Thanks for helping me out. You're an awesome person and I hope you have a great day!

thisiskeithb, 2022-05-01:
I put in PRs https://github.com/MarlinFirmware/Marlin/pull/24120 and and https://github.com/MarlinFirmware/Configurations/pull/726 to address this.

closed 2022-05-01 by thisiskeithb