MarlinFirmware/Marlin #24124: [BUG] (bug summary)

RexfordProgrammer, 2022-05-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Reproduction Steps:

Using STM32F103RC_btt environment with SKR Mini e3 V2,

Or using a ?different environment (one which I cannot remember) for Creality Ender 3 Pro 4.2.7 "V2?" board

on Marlin Version SHORT_BUILD_VERSION "2.0.9.3"

When BLTOUCH is enabled, with BL Touch 3.1, No adapter board, Z Endstop is inverted by default.

### Bug Timeline

Began on upgrade from 4.2.2 to 4.2.7 board

### Expected behavior

I expected the probe to deploy and the z axis would move until it was stowed by touching the bed

### Actual behavior

Probe would deploy and stow several times and z axis wouldn't move downward until probe was depressed.

### Steps to Reproduce

1. Enable BL Touch
2. Z_MIN_PROBE_ENDSTOP_INVERTING

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ender 3 Pro

### Electronics

SKR Mini E3 V.2 or Ender 3 Pro 4.2.7 Board

### Add-ons

BLTouch 3.1 No Adapter Board

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Additional information & file uploads

the line below must be set to true for BLTouch to work
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe.

By editing Conditionals_LCD.h, I was able to disable the safegaurd, and enable this behavior, thus solving the problem of an inverted probe behavior.

  // Always disable probe pin inverting for BLTouch
  #if Z_MIN_PROBE_ENDSTOP_INVERTING
    //#error "BLTOUCH requires Z_MIN_PROBE_ENDSTOP_INVERTING set to false. Please update your Configuration.h file."
  #endif
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/8603024/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/8603026/Configuration.txt)
[Conditionals_LCD.txt](https://github.com/MarlinFirmware/Marlin/files/8603038/Conditionals_LCD.txt)

ellensp, 2022-05-02:
a real requires Z_MIN_PROBE_ENDSTOP_INVERTING set to false, always
Enable BLTOUCH_FORCE_SW_MODE

RexfordProgrammer, 2022-05-02:
> 

I rebuilt as requested, this did not resolve my issue.

I set, Z_MIN_PROBE_ENDSTOP_INVERTING to false
uncommented BLTOUCH_FORCE_SW_MODE 

Yes, I do have a real BLTOUCH 3.1, Creality packaging and the whole 9 yards. Cost me a pretty penny, and is not a knockoff. 

But I do appreciate you trying to help.

andyspoo, 2022-05-15:
I have exactly the same issue and I can confirm that it is because the compiler checks will not allow =

#define Z_MIN_PROBE_ENDSTOP_INVERTING true

This seems to be the code in Conditionals_LCD.h that is preventing it =

  // Always disable probe pin inverting for BLTouch
  #if Z_MIN_PROBE_ENDSTOP_INVERTING
    #error "BLTOUCH requires Z_MIN_PROBE_ENDSTOP_INVERTING set to false. Please update your Configuration.h file."
  #endif

  #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
    #if Z_MIN_ENDSTOP_INVERTING
      #error "BLTOUCH requires Z_MIN_ENDSTOP_INVERTING set to false. Please update your Configuration.h file."
    #endif
  #endif
#endif

andyspoo, 2022-05-15:
Just to confirm that I'm using the latest official release of Marlin and the bltouch was working perfect using a Marlin version from about 2 years ago, a version of Marlin that allowed "#define Z_MIN_PROBE_ENDSTOP_INVERTING true" just fine.
I'm using a Bigtreetech SKR Mini E3 V1.2

RexfordProgrammer, 2022-05-15:
@andyspoo 

You did comment out that line in CONDITIONALS_LCD.h to correct the issue right? It isn't a difficult change.

andyspoo, 2022-05-16:
I'm no coder, but I disabled both the error lines in CONDITIONALS_LCD.h that began with #error using  // 
but it still failed.
Any suggestions?
Thanks.

thisiskeithb, 2022-05-16:
> Just to confirm that I'm using the latest official release of Marlin and the bltouch was working perfect using a Marlin version from about 2 years ago, a version of Marlin that allowed "#define Z_MIN_PROBE_ENDSTOP_INVERTING true" just fine. I'm using a Bigtreetech SKR Mini E3 V1.2

Because Marlin was silently fixing it for you. It no longer does that and now gives an error.

As Dust mentioned above:

> a real requires Z_MIN_PROBE_ENDSTOP_INVERTING set to false, always, no exceptions, ever.
Enable BLTOUCH_FORCE_SW_MODE

RexfordProgrammer, 2022-05-18:
> I'm no coder, but I disabled both the error lines in CONDITIONALS_LCD.h that began with #error using // but it still failed. Any suggestions? Thanks.

You may need to clean your build. You will be able to make that variable false if the proper check listed in the terminal is disabled. I can definitely help more directly if need be.

thisiskeithb, 2022-05-18:
Stop disabling the error lines. They are there for a reason.

While you think you are tricking Marlin to get back to how it was before, Marlin was silently fixing the incorrect `Z_MIN_PROBE_ENDSTOP_INVERTING ` setting in previous versions.

*It no longer does that and errors out instead.*

A BLTouch requires `Z_MIN_PROBE_ENDSTOP_INVERTING` set to false, always, no exceptions, ever.

closed 2022-05-18 by thisiskeithb

RexfordProgrammer, 2022-05-18:
Why is this closed with no resolution.

thisiskeithb, 2022-05-18:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

RexfordProgrammer, 2022-05-18:
I have already sought help from the community, this is a last ditch effort to get this resolved without doing the override I was forced to concoct. I will continue to do it this way, if need be. But I don't see why is this is not a bug if it isn't with the firmware I had to "trick".

Keep in mind. It works. And only the way I describe. I don't need help. I need an understanding of which my legitimate bl touch 3.1 requires inversion. And why I shouldn't do it this way, if this works.