MarlinFirmware/Marlin #24127: [BUG] MEDIA INIT FAIL

Noobje130, 2022-05-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

when flashing marlin or marling bugfix to my biqu b1 se  plus skr 2B evreything works but cant intalize sd card

### Bug Timeline

when using marlin

### Expected behavior

to be able to reead the sdcard

### Actual behavior

cant read the sdcard

### Steps to Reproduce

1.  download latest marlin bugfix
2. download configuration example use biqu b1 se plus
3. set default env to BIGTREE_SKR_2
4. error

### Version of Marlin Firmware

bugfix 2.0.x

### Printer model

biqu b1 se plus

### Electronics

stock

### Additional information & file uploads

[image]
[config.zip](https://github.com/MarlinFirmware/Marlin/files/8621436/config.zip)

ellensp, 2022-05-04:
You enabled usb  ie   #define USB_FLASH_DRIVE_SUPPORT, did you plug in a usb stick?

Noobje130, 2022-05-10:
@thisiskeithb my sd card reader is working now but i cant use my usb port 

what i did to get my sd card reader working was disable USB_FLASH_DRIVE_SUPPORT
then i got these errors (look at pictures) to fix i did what it sayed change to DiskIODriver_SDIO

but i wonder is there way my usb port could work too

Noobje130, 2022-05-10:
and btw me tried enabling USB_FLASH_DRIVE_SUPPORT again ofc

thisiskeithb, 2022-05-10:
With `USB_FLASH_DRIVE_SUPPORT` disabled, you lose access to that port.

Having said that, I've confirmed here that with `MULTI_VOLUME` enabled, you can select which one you want to print from.

Noobje130, 2022-05-10:
@thisiskeithb even with USB_FLASH_DRIVE_SUPPORT enabled and MULTI_VOLUME too me can print from sd card no problem but it does not detect my usb drive

Noobje130, 2022-05-10:
like it gives menu to select sd or sb but it shows the same for both

thisiskeithb, 2022-05-10:
Try power cycling the printer with both plugged in.

APToni, 2022-11-24:
I ran into a similar problem on the mks robin nano v3 board, with a 12864B display without a resistor. With the normal marlin basic setting, the sd card works fine, but when adding a usb drive (in conf marlin), a similar message appears, but I can select a file to print with the usb flash drive installed from both the sd card and the usb drive. BUT I can not open the SD card WITHOUT usb drive. Here is a feature in the firmware.

trengtor, 2022-12-25:
I was upgrade Marlin fron 2.0.8.2 to 2.1.2 on contol board BTT E3 RRF V1.1 — now have problem with media init fail.

trengtor, 2022-12-25:
https://github.com/MarlinFirmware/Marlin/issues/25151

reynum2, 2023-06-02:
I have the same problem with ramps 1.4 on a printer Anet A8 with Marlin 2.0.9.6

I have enable 
#define SDSUPPORT
#define SDCARD_CONNECTION LCD

And tried with 
#define MULTI_VOLUME
#define USB_FLASH_DRIVE_SUPPORT

I wanted to test with 
#define SDIO_SUPPORT 
But the compilation fail with this parameter

I use the SD card reader in the screen like this one :
[https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller](https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller)
and activate : 
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

My card is 1Go an formated in FAT32 but I still have the "Media init fail" message when I insert the card

EvilGremlin, 2023-06-02:
because SDIO doesn't exist on AVR
USB_FLASH_DRIVE_SUPPORT - do you even have this hardware?

reynum2, 2023-06-04:
I tried with 4 cards and the result is the same.
I will try to update to last Marlin version and if it does not work : create a new bug

reynum2, 2023-06-04:
try with Marlin 2.1.2.1 and still the : 
[image]

ellensp, 2023-06-05:
USB_FLASH_DRIVE_SUPPORT requires an additional board, not a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

```
   /**
   * Support for USB thumb drives using an Arduino USB Host Shield or
   * equivalent MAX3421E breakout board. The USB thumb drive will appear
   * to Marlin as an SD card.
   *
   * The MAX3421E can be assigned the same pins as the SD card reader, with
   * the following pin mapping:
   *
   *    SCLK, MOSI, MISO --> SCLK, MOSI, MISO
   *    INT              --> SD_DETECT_PIN [1]
   *    SS               --> SDSS
   *
   * [1] On AVR an interrupt-capable pin is best for UHS3 compatibility.
   */
  //#define USB_FLASH_DRIVE_SUPPORT
```

reynum2, 2023-06-05:
Yes I tried with this parameter (USB_FLASH_DRIVE_SUPPORT) but remove it when I saw it does not work.

Now the only parameter related to the SD is 
`
#define SDSUPPORT
`

redskin61, 2023-08-11:
I resolve media init fail after commenting the line "//#define SD_DETECT_STATE HIGH" in Configuration.adv.h--- SKR PRO v1.1

aurion55, 2024-04-26:
> Yes I tried with this parameter (USB_FLASH_DRIVE_SUPPORT) but remove it when I saw it does not work.
> 
> Now the only parameter related to the SD is `#define SDSUPPORT` But maybe there is another parameter to enable ?

Did you look here:
 SD Card SPI Speed
   * May be required to resolve "volume init" errors.
   *
   * Enable and set to SPI_HALF_SPEED, SPI_QUARTER_SPEED, or SPI_EIGHTH_SPEED
   *  otherwise full speed will be applied.
   *
   * :['SPI_HALF_SPEED', 'SPI_QUARTER_SPEED', 'SPI_EIGHTH_SPEED']
   */
  //#define SD_SPI_SPEED SPI_HALF_SPEED
  
  
  It state above if you have "Volume init error"you might need to low to half speed?
  
  I'm on Marlin 2.1.2 and just started to get these "media init error" using the same setup as you, I think to due to dirty pins on the Sd card? As I never has in the past.

aurion55, 2024-04-26:
Clean your Sd Card with some alcohol wipes, as I'm going to do the same.

closed 2024-09-24 by github-actions[bot]