MarlinFirmware/Marlin #24131: [BUG]

dmitrygribenchuk, 2022-05-04:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Hello. Have a firmware crash when inserting media and scroll at the same time

https://user-images.githubusercontent.com/15670247/166807142-b926a798-2e1b-46fb-8fd0-80ad397237cb.mp4

### Steps to Reproduce

1) Insert or remove media
2) Scroll a button

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ender 3v2 (4.2.2)

ellensp, 2022-05-05:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

dmitrygribenchuk, 2022-05-05:
Try to reproduce it on [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) it seems that need a little bit more time for that but still able.

ellensp, 2022-05-05:
Since you seem to have the knack on triggering this, could you add #define WATCHDOG_DURATION_8S to the end of your your Configuration.h recompile and upload and see if the issue still occurs?

dmitrygribenchuk, 2022-05-05:
I'm adding #define WATCHDOG_DURATION_8S to the end of my Configuration.h recompile and upload and the issue still occurs

ellensp, 2022-05-05:
The watchdog is a timer that must be reset before the the time limit or it reboots. This is how you detect and get out of software crashes.
That define doubles that time from 4 to 8 seconds.  Since it made no difference. It probably not tripping the watchdog timer.

ellensp, 2022-05-05:
It doesn't work that ways, Is used in code like this

`#define WDT_TIMEOUT_US TERN(WATCHDOG_DURATION_8S, 8000000, 8000000) // 4 or 8 second timeout
`

So if WATCHDOG_DURATION_8S is set use 8000000, else use 8000000

Other WATCHDOG_DURATION_ values are not checked for

mrbass21, 2022-06-26:
I don't know if it's related, but I've triggered a similar situation. I'm brand new to 3D printing and have an Ender 3 Pro (4.2.2 board, all stock for now). I started messing with upgrading Marlin late last night. I went from 2.0.8.2 to 2.1 and while printing, I constantly get beeps from the piezo and the display goes into a menu as though I pushed in on my button.

While I was printing I went to the Power Outage setting and clicked in on it, the software rebooted (crashed?) and the print was immediately stopped.

I'm still looking through tickets to see if anyone has the problem I do with the logic board constantly thinking I've interacted with the menu during a print. If I let it sit, sometimes it goes back through phantom selection, and sometimes it times out and goes away. Seemed maybe tangentially related. I'll keep looking at issues in the mean time to see if anyone else is having my problem.

dmitrygribenchuk, 2022-08-26:
Still can reproduce on 2.1.1

thisiskeithb, 2022-08-26:
Since https://github.com/MarlinFirmware/Marlin/pull/24652 was patched a few days ago, please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

dmitrygribenchuk, 2022-08-26:
Also reproduced on bugfix-2.1.x

closed 2022-11-06 by github-actions[bot]

dmitrygribenchuk, 2022-11-06:
Still reproduced on bugfix-2.1.x