MarlinFirmware/Marlin #24133: Error: Unknown environment names 'lpc1768'

Maxzprint, 2022-05-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When attempting to build 2.0.9.3 in platformio using LPC1768 environment vscode reports the following error "Error: Unknown environment names 'lpc1768'"
No issue building in earlier versions such as 2.0.9.1

### Bug Timeline

I encountered this issue 6th May 2022

### Expected behavior

I expect firmware to build

### Actual behavior

Build fails with error

### Steps to Reproduce

compile firmware in vscode
select LPC1768 build environment
attempt build
[2.0.9.3.zip](https://github.com/MarlinFirmware/Marlin/files/8637348/2.0.9.3.zip)

### Version of Marlin Firmware

2.0.9.3

### Printer model

irrelavant

### Electronics

Big Tree Tech SKR 1.3

thisiskeithb, 2022-05-06:
Use `LPC1768`, not `lpc1768`.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-05-06 by thisiskeithb

Maxzprint, 2022-05-06:
Answer is irrelavant, error report is as presented by vscode, I did not type lpc1768
Also this is a bug report as I,m not seeking config assistance, it seems the LPC1768 environment will not build using 2.0.9.3

ellensp, 2022-05-06:
The only way to get this error is to enter default_envs = lpc1768  vs default_envs = LPC1768 in the platfformio.ini file
Or running pio from cmd line  pio run -e lpc1768 vs  pio run -e LPC1768

Maxzprint, 2022-05-06:
Ok, This configuration will not build on 2.0.9.3 but the same builds on 2.0.9.1
Perhaps I'm not understanding something but it seems to be an issue confined to 2.0.9.3

ellensp, 2022-05-06:
Provided config files are not correct and contain error: #error "ADVANCED_PAUSE_FEATURE requires NOZZLE_PARK_FEATURE." This is a user configuration issue.

Maxzprint, 2022-05-06:
Here is the actual error, the avanced pause error was corrected once I was able to move past this point in 2.0.9.1.
[image]

ellensp, 2022-05-06:
The error is exceedingly clear.

you have lpc1768 in your platformio.ini 

Show your platformio.ini file, line 16

Maxzprint, 2022-05-06:
Thank you I wasn't looking at platformio.