MarlinFirmware/Marlin #24134: CRTouch and Creality Spider Hotend support (both on one firmware file)

r0dnay, 2022-05-06:
### Describe the feature you want

provide support for Creality Spider Hotend with new barrel type thermistor (500°C capable hotend) and also crtouch in one firmware

creality have firmware for each but not for both - very odd

ellensp, 2022-05-06:
Creality do not provide any documentation for this hotend and it is not in the source they say if for these firmwares.

r0dnay, 2022-05-06:
thanks for the response although im not sure i can make sense of your reply

> and it is not in the source they say is for these firmwares.

i know there isnt any documentation other than installation guides, not sure what the rest of your post is trying to say

ellensp, 2022-05-06:
The code would have done in lieu of documentation. 
Without it we have zero to go on..
I looked through the ender 3 high temp binary, but could only find a standard type 1 thermistor table.

ellensp, 2022-05-06:
this install guide

[image]

Shows you reuse the stock hotend heater and thermistor. 

This cannot do > 300C

r0dnay, 2022-05-06:
the thermistor is different , the spider now comes bundled with a barrel type thermistor and the instructions call for the stock glass thermistor be removed and replaced, i believe this is why its rated at 500

however, the firmware on creality's site, changed for the spider hotend only goes up to 300, which is only 40 over stock, which considering whats advertised, is pretty lacklustre, and i have a feeling it uses the same table

the spider has been out for a little while, surely theres some info we can get our hands on regarding tables, specs etc? do any of the marlin devs have a contact at creality? it would essentially benefit them too as currently their firmware support for the spider looks inadequate and doesnt allow the hotend to be used at its advertised potential.

also..is there a way to read the code from both crtouch firmware bin and also the firmware listed to accomodate the spider hotend and consolidate the differentials? currently seems to be a crtouch firmware and a seperate hotend firmware, and the clever guys at creality havnt yet thought of putting both into one, just in case customers have upgraded both on one machine

i actually got a response from official creality store on amazon who say they havnt yet made firmware which facilitates both hardware upgrades

InsanityAutomation, 2022-05-07:
I've got one and the temp table appears to line up to the type 1 300c unit. It's not uncommon to have the hotend itself rated for more than the supplied thermistor. Most likely if you want to go that high, you'll need to add a pt1000.

thisiskeithb, 2022-05-07:
> I've got one and the temp table appears to line up to the type 1 300c unit. It's not uncommon to have the hotend itself rated for more than the supplied thermistor. Most likely if you want to go that high, you'll need to add a pt1000.

OP can use `BLTOUCH` for CRTouch support, so there’s nothing else to add to Marlin.

closed 2022-05-07 by thisiskeithb

r0dnay, 2022-05-07:
> > I've got one and the temp table appears to line up to the type 1 300c unit. It's not uncommon to have the hotend itself rated for more than the supplied thermistor. Most likely if you want to go that high, you'll need to add a pt1000.

when you say you have got one, thats the packaged upgraded thermistor or spider hotend?

> 
> OP can use `BLTOUCH` for CRTouch support, so there’s nothing else to add to Marlin.

so if i was to use marlin, the current stable release would cater for both crtouch (i know you can use any ABL with 'BLTouch') and also the upgraded hotend and thermistor?

r0dnay, 2022-05-07:
also, i got a message from creality on amazon who say the upgraded thermistor is actually a pt100 which is capable of 500°C (not a typo, its not the same as the pt1000)

thisiskeithb, 2022-05-07:
> so if i was to use marlin, the current stable release would cater for both crtouch

There's nothing special to add to Marlin since PT100 RTDs and BLTouch/clones are already supported in Marlin and have been for a long time.

r0dnay, 2022-05-07:
brill, so if the resistance table for the pt100 exists i should be able to configure marlin accordingly and get the temps up to 500 - ill do that, thanks for clarifying

r0dnay, 2022-05-09:
just wanted to leave a comment after recieving a message from Creality's engineers, so if anyone else has a Spider hotend, or thinking of getting one is fully aware.

the staff member who replied to my message wasnt aware that the pt100 isnt shipped with the spider, its a stainless steel 300 degree thermistor which the engineers at Creality have confirmed.

how the information between the Creality design department and marketing department got 'confused' is beyond me.

the creality UK store on amazon have said they will be changing the description of the item to reflect maximum temperature of 300 degrees, instead of 500

wether this means dont upgrade the thermistor or that the heating block isnt capable of 500 degrees - i dont know, but the fact that they are now changing the 500 to 300 instead of advising that the thermistor can be upgraded to a PT type etc makes me think the heating block should stay around the 300 degrees mark.
otherwise they would have simply told me to upgrade thermistor if i want to reach 500 degrees celcius.