MarlinFirmware/Marlin #24135: [BUG] Copy past error in Endstops.cpp

rondlh, 2022-05-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I just noticed some odd code in endstop.cpp, MARKFORGED_XY is check twice.
I suspect it's a copy and past error and it should be MARKFORGED_XY, MARKFORGED_YX.

Line 620:   #if ENABLED(G38_PROBE_TARGET) && NONE(CORE_IS_XY, CORE_IS_XZ, MARKFORGED_XY, MARKFORGED_XY)
Line 631:   #if ANY(CORE_IS_XY, CORE_IS_XZ, MARKFORGED_XY, MARKFORGED_XY)

### Bug Timeline

Don't know

### Expected behavior

Odd code

### Actual behavior

Logical code

### Steps to Reproduce

Check endstops.cpp, line 620 and 631.

### Version of Marlin Firmware

Marlin 2.0.9.3 latest bugfix

### Printer model

Custom

### Electronics

Custom

### Add-ons

BTT TFT35 V3.0

### Bed Leveling

ABL Bilinear mesh

### Host Software

Cura

### Additional information & file uploads

Perhaps it's nothing, but it looks odd...

ellensp, 2022-05-06:
agreed

was added https://github.com/MarlinFirmware/Marlin/commit/e5154ec2817  

which was meant to add MARKFORGED_YX

ellensp, 2022-05-07:
Pr created to fix this, closing

closed 2022-05-07 by ellensp