MarlinFirmware/Marlin #24137: [BUG] (bug summary)mar laser

bihailantian655, 2022-05-06:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

while  using  lightburn   G1 X10 Y10 S999  ,the  max pwm  duty  is only  90percent ,    not  100percent 

    #define LASER_TEST_PULSE_MIN           1   // Used with Laser Control Menu
    #define LASER_TEST_PULSE_MAX         999   // Caution: Menu may not show more than 3 characters

#define LASER_POWER_INLINE

    #if ENABLED(LASER_POWER_INLINE)
    #define LASER_POWER_INLINE_TRAPEZOID
   #define LASER_POWER_INLINE_TRAPEZOID_CONT
  #define LASER_POWER_INLINE_TRAPEZOID_CONT_PER 20
   #define LASER_MOVE_POWER

      #if ENABLED(LASER_MOVE_POWER)
        #define LASER_MOVE_G0_OFF
        #define LASER_MOVE_G28_OFF
      #endif

### Bug Timeline

no

### Version of Marlin Firmware

2.0.9

### Printer model

esp32 

### Electronics

esp32

### Add-ons

no

### Additional information & file uploads

no

ellensp, 2022-05-07:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

bihailantian655, 2022-05-07:
i  have test      [bugfix-2.0.x branch]    problem  still exist

bihailantian655, 2022-05-07:
parts of  my   laser   config  

      #define LASER_FEATURE
      #define SPINDLE_LASER_ACTIVE_STATE    LOW   // Set to "HIGH" if SPINDLE_LASER_ENA_PIN is active HIGH
      // changed    defaugt low 
      // #define SPINDLE_LASER_FREQUENCY     2500   // (Hz) Spindle/laser frequency (only on supported HALs: AVR and LPC)
     #define SPINDLE_LASER_ENA_PIN              2  // Pullup or pulldown!
     #define SPINDLE_LASER_PWM_PIN              0  // Hardware PWM
     #define LASER_POWER_INLINE
     #define LASER_POWER_INLINE_TRAPEZOID_CONT
     #define LASER_POWER_INLINE_TRAPEZOID_CONT_PER 20
     #define LASER_MOVE_POWER
     #define LASER_MOVE_G0_OFF
     #define LASER_MOVE_G28_OFF

descipher, 2022-05-10:
See #22690 for some laser updates -> https://github.com/descipher/Marlin/archive/refs/heads/laser.gcode.inline.fix.zip
use this for configuring -> https://marlinfw.org/docs/configuration/2.0.9/laser_spindle.html
[image]
[image]

Use percent if you reference power that way.
Configuration_adv.h -> #define CUTTER_POWER_UNIT PERCENT

bihailantian655, 2022-05-11:
after modified Configuration_adv.h      Configuration.h       esp32 pin  platform.ini   according to my previous  settings   then 
 compiled  the    laser.gcode.inline.fix    

 dont  know  why  now  there is  no pwm generation  on the pwm pin .  

steppers  working  normal

HoverClub, 2022-05-17:
You need to define ```SPINDLE_LASER_FREQUENCY``` otherwise hardware PWM won't work.

bihailantian655, 2022-05-17:
I have tried uncommenting   define SPINDLE_LASER_FREQUENCY  , but the result was the same

descipher, 2022-05-17:
Please upload the config files.

bihailantian655, 2022-05-17:
[bugfix_config.zip](https://github.com/MarlinFirmware/Marlin/files/8710429/bugfix_config.zip)

Thank you for your attention. This is my document. I don't know where the configuration is wrong

descipher, 2022-05-17:
You are using the wrong version of configuration files for the PR, it needs 02010000 to work correctly.
You should start with the config files in the PR and change only the existing values as required.
There seems to be a number of unusual values in the files. e.g. HEATER_0_MINTEMP   -5
If unknown, leave it as default.
I would suggest you only edit the very minimum to start.
Only enable LASER_FEATURE first then add options after testing basics.

HoverClub, 2022-05-17:
Also ```CUTTER_POWER_UNIT PWM255``` and using ```G1 X10 Y10 S999``` will set the wrong power (the range is set to  0-255 - better/easier to use ```CUTTER_POWER_UNIT PERCENT``` and set it in Lightburn?).

```LASER_TEST_PULSE_MAX``` is the test pulse time (mS) not the maximum laser value (which is set by ```CUTTER_POWER_UNIT```).

bihailantian655, 2022-05-18:
> You are using the wrong version of configuration files for the PR, it needs 02010000 to work correctly. You should start with the config files in the PR and change only the existing values as [...]

I made these changes for the following reasons

1   previous is  master branch 2.09    not  bugfix    

2   HEATER_0_MINTEMP -5     ：This machine is not used for printing   just  for laser        （
我原以为温度设置太高会不正常

I thought the temperature setting too high would be abnormal）

thisiskeithb, 2022-05-18:
> I made these changes for the following reasons

> 1 previous is master branch 2.09 not bugfix

PRs are against `bugfix-2.0.x`, so you need to make your changes as if you were using that branch.

bihailantian655, 2022-05-18:
> > I made these changes for the following reasons
> 
> > 1 previous is master branch 2.09 not bugfix
> 
> PRs are against `bugfix-2.0.x`, so you need to make your changes as if you were using that branch.

Under the root directory of the compressed package is the  files  of  bugfix

bihailantian655, 2022-05-18:
these are all  bugfix confix
[Uploading all  bugfix_files.zip…]()

I'll upload the file again

descipher, 2022-05-18:
My bad, I was looking at the previous folder.

bihailantian655, 2022-05-18:
> Also `CUTTER_POWER_UNIT PWM255` and using `G1 X10 Y10 S999` will set the wrong power (the range is set to 0-255 - better/easier to use `CUTTER_POWER_UNIT PERCENT` and set it in Lightburn?). [...]

Thank you. I misunderstood the function of configuration，  
 LASER_TEST_PULSE_MAX         999        CUTTER_POWER_UNIT PWM255
I thought the maximum was 1000,   so  lightburn  set  max s value    1000  ，it  should  be 255 。

descipher, 2022-05-18:
Lightburn and LaserGRBL would use the 1000 default.

I think this line if the issue in laser_spindle.cpp:82
    TERN_(MARLIN_DEV_MODE, frequency = SPINDLE_LASER_FREQUENCY);

Try enabling MARLIN_DEV_MODE, it not required any more and should just be frequency = SPINDLE_LASER_FREQUENCY
But now it is multiple locations.
Seems to have snuck in with the HAL refactor.

bihailantian655, 2022-05-18:
Thank you for answering and solving my question. I'm not a developer. Now everything is normal. I'll use the stable version in the future

bihailantian655, 2022-05-18:
> Lightburn and LaserGRBL would use the 1000 default.
> 
> I think this line is the issue in laser_spindle.cpp:82 TERN_(MARLIN_DEV_MODE, frequency = SPINDLE_LASER_FREQUENCY);
> 
> Try enabling MARLIN_DEV_MODE, it is not required any more and should just be frequency = SPINDLE_LASER_FREQUENCY But now it is multiple locations. Seems to have snuck in with the HAL refactor.

thank you  ，

closed 2022-05-18 by bihailantian655