MarlinFirmware/Marlin #24138: [BUG] /u8g_fontutf8.h:11:24: fatal error: U8glib-HAL.h: No such file or directory

countcobolt, 2022-05-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to compile the latest bugfix-2.0.0  (git cloned) for my MKS Tinybee mainboard.
I am working with platformio from within commandline. When trying to compile, it failed with the following error

```
In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/../../../../../Marlin/src/lcd/lcdprint.h:53:0,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/../../../../../Marlin/src/lcd/marlinui.h:80,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/espcom.cpp:36:
.pio/libdeps/mks_tinybee/ESP3DLib/src/../../../../../Marlin/src/lcd/dogm/u8g_fontutf8.h:11:24: fatal error: U8glib-HAL.h: No such file or directory

```
Yet, when I do a platformio lib list, you can see the library installed at the bottom

```
Library Storage: /home/countcobolt/firmwares/TinyBee/Marlin/.pio/libdeps/mks_tinybee
AsyncTCP
========
Asynchronous TCP Library for ESP32

Version: 1.1.1
License: LGPL-3.0
Keywords: async, tcp
Compatible frameworks: arduino
Compatible platforms: espressif32
Authors: Hristo Gochkov

ESP Async WebServer
===================
Asynchronous HTTP and WebSocket Server Library for ESP8266 and ESP32

Version: 1.2.3
License: LGPL-3.0
Keywords: http, async, websocket, webserver
Compatible frameworks: arduino
Compatible platforms: espressif8266, espressif32
Authors: Hristo Gochkov

ESP32SSDP
=========
Simple SSDP library for ESP32. Only for ESP32

Version: 1.2.0
Keywords: communication
Compatible frameworks: arduino
Compatible platforms: espressif32
Authors: Me-No-Dev, luc-github

ESP3DLib
========
A 3D printer front end for ESP boards. This library implements a 3D printer front end.

Version: 1.0.9
Keywords: communication
Compatible frameworks: arduino
Compatible platforms: espressif32
Authors: Luc Lebosse
Source: https://github.com/luc-github/ESP3DLib/archive/master-2.0.7.zip

ESPAsyncTCP
===========
Asynchronous TCP Library for ESP8266

Version: 1.2.2
License: LGPL-3.0
Keywords: async, tcp
Compatible frameworks: arduino
Compatible platforms: espressif8266
Authors: Hristo Gochkov

U8glib-HAL
==========
#ID: 1932
Unofficial repository for combined U8G and U8Glib-ARM with HAL extensions

Version: 0.5.2
Keywords: u8g, u8glib, arm, hal
Compatible frameworks: *
Compatible platforms: *
Authors: Scott Lahteine

WebSockets
==========
WebSocket Server and Client for Arduino based on RFC6455

Version: 2.3.4
License: LGPL-2.1
Keywords: wifi, http, web, server, client, websocket
Compatible frameworks: arduino
Compatible platforms: atmelavr, espressif8266, espressif32
Authors: Markus Sattler

```

### Bug Timeline

New for me

### Expected behavior

Compilation that finished

### Actual behavior

doesn't compile

### Steps to Reproduce

#git clone https://github.com/MarlinFirmware/Marlin.git
#cd Marlin
#git switch bugfix-2.0.x
#git fetch
 modify filament sensor pin in  src/pins/esp32/pins_MKS_TINYBEE.h
 Copy config over.
#platformio run -e mks_tinybee

### Version of Marlin Firmware

not on printer yet

### Printer model

CR10 heavy modified

### Electronics

MKS Tinybee board

### Add-ons

-

### Bed Leveling

ABL 3-point

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/8642441/config.zip)

ellensp, 2022-05-07:
That is a 3rd party library ESP3DLib, not part of Marlin.

ellensp, 2022-05-07:
In saying that I have already fixed this error in the library, but marlin is looking at old version of the library

In ini/features.ini  is

```
(ESP3D_)?WIFISUPPORT                   = AsyncTCP, ESP Async WebServer
                                         ESP3DLib=https://github.com/luc-github/ESP3DLib/archive/master-2.0.7.zip
                                         arduinoWebSockets=links2004/WebSockets@2.3.4
                                         luc-github/ESP32SSDP@^1.1.1
                                         lib_ignore=ESPAsyncTCP

```
Update ESP3DLib to 
                                         ESP3DLib=https://github.com/luc-github/ESP3DLib/archive/refs/heads/master.zip

ellensp, 2022-05-07:
Closing as PR has been created.

closed 2022-05-07 by ellensp

countcobolt, 2022-05-07:
thank you very much!