MarlinFirmware/Marlin #24139: [BUG] Z Endstop not triggering at MKS BASE/GEN L v2.1 #16393[BUG] (bug summary)

Slamar97, 2022-05-06:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Hello, 

my issue is related to #16393, at the first time (few months ago) this case solved my problem, but 2 days ago when I want to add mesh surface with  manual levelling. I opened my "working" config and uncomment #define PROBE_MANUALLY
 and #define MESH_BED_LEVELING and #define LCD_BED_LEVELING I compiled and upload new firmware and 😞 z axis end stop not working. I thought there is sth wrong with mesh  and I decided to get back without it (I commented above lines) and nothing changed, still not working. 

Today I decided to set up again all config from scratch. To make it I used config example of Ender 3 Pro 422 board. Old and new one was made on 2.0.9.3. Result,  nothing changed. Still now working. (Limit switch hardware working I checked it with multimeter)  I noticed one thing before I disable Serial 2, Y axis end stop also not works, after comment serial 2 works but z axis end stop no. Second problem become with new firmware, the console over USB port not working. Ex I'm trying to connect via console in Repeatier host it connecting, but there is no communication. I guess it's because of disabled serial 2, but before upgrade firmware i connected succesfully and make a backup of parameters in eeprom.

my hardware:
Creality Ender 3 PRO
Markerbase MKS Gen L V2.1
TMC2209 fystec
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8643769/Marlin.zip)

Thank you for your time
Best 
Marcin

### Bug Timeline

new

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ender 3 PRO

### Electronics

Markerbase MKS Gen L V2.1 with TMC2209 fystec

Zatarines, 2022-05-08:
I am having the same exactly issue w/ Z_Endstop not working  MKS Gen L V2.1. Axis X n dual Y working fine but the Z_axis does not move all the way down to trigger endstop. Upon HomeZ command, Z moves down a little bit, bumps midair and goes down a little as if it had hit z_min endstop. If keep sending the Home command it will keep moving the same not using endstop. I checked endstop Triggers status w/ M119 Z status is always open no matter if stop is mechanically pressed. all connections checked to be okay.
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/8646345/Archive.zip)

ellensp, 2022-05-08:
@Zatarines try enabling ENDSTOP_NOISE_THRESHOLD

Zatarines, 2022-05-08:
@ellensp thank you for reply.. that didnt work though.. Everything works fine, but that Zhoming..

ellensp, 2022-05-08:
@Zatarines  @Slamar97 
You have #define SERIAL_PORT 1 In your Configuration.h
Serial port 1 uses pins D18 and D19
Z_MIN_PIN is on D18 Z_MAX_PIN is on D19
This will never work.

ellensp, 2022-05-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-05-08 by ellensp

Zatarines, 2022-05-08:
@ellensp thank you for your support. what is the pathway for one to acquire skills in troubleshooting and writing FW like this?

Slamar97, 2022-05-09:
Thanks for your answers, I know there is conflict in pins, but I'm wondering how it worked before for 2 months, without any issue

Slamar97, 2022-05-09:
@Zatarines I solved my problem with Z end stop by redefine in configuration.h
#define SERIAL_PORT 1
to 
#define SERIAL_PORT 0

Zatarines, 2022-05-10:
@Slamar97  thank you, that fixed my issue also.