MarlinFirmware/Marlin #24142: [BUG] MKS TinyBee wrong temp reported on 2.0.9.3 & Bugfix

countcobolt, 2022-05-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The firmware is reporting wrong temperatures of the thermistors. On the original MKS version (https://github.com/makerbase-mks/MKS-TinyBee/tree/main/firmware/mks%20tinybee%20marlin) , I get a proper read out of the temperature (in °C, 21 bed,  27 hotend). 

When building either Bugfix or 2.0.9.3 I get wrong reading. Bugfix is reading 0 and 1°C for both bed and hotend, while the 2.0.9.3 is reading a lot higher temperature around 70°C.

Now, when I revert back to the MKS version, the temperatures are back correct, indicating that this is not a thermistor or machine error. 

I have updated the pins_MKS_TINYBEE.h but only tried to add the ADC reading and the filament out pin. 

No other changes to the 2.0.9.3 version or bugfix.

### Bug Timeline

new

### Expected behavior

The right temperature to show on all firmware versions

### Actual behavior

Wrong temp on bugfix and 2.0.9.3

### Steps to Reproduce

Install MKS version of  firmware (2.0.9.2 branch)
Check temperature
Build either bugfix or 2.0.9.3 
Check temperature
Difference is hug
[MKS_Original.zip](https://github.com/MarlinFirmware/Marlin/files/8644638/MKS_Original.zip)
[20903_config.zip](https://github.com/MarlinFirmware/Marlin/files/8644639/20903_config.zip)
e

### Version of Marlin Firmware

2.0.9.2 MKS version

### Printer model

CR10 heavy modified

### Electronics

MKS Tinybee board

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

I have upload the original config that is working (MKS_Original, this does contain the changes for my machine) and the 2.0.9.3 configuration that is showing the wrong version. Bugfix I did not keep but can provide if needed.

countcobolt, 2022-05-07:
Solved by using solution in here https://github.com/makerbase-mks/MKS-TinyBee/issues/7

closed 2022-05-07 by countcobolt

ellensp, 2022-05-07:
That is already in bugfix... 

with this in Marlin/src/pins/esp32/pins_MKS_TINYBEE.h

```
//
// ADC Reference Voltage
//
#define ADC_REFERENCE_VOLTAGE                2.5  // 2.5V reference VDDA

```

and ths is the HAL

```
#ifndef ADC_REFERENCE_VOLTAGE
  #define ADC_REFERENCE_VOLTAGE 3.3
#endif
```
ie set to 2.5 for MKS_TINYBEE

then used here

```
void MarlinHAL::adc_start(const pin_t pin) {
  const adc1_channel_t chan = get_channel(pin);
  uint32_t mv;
  esp_adc_cal_get_voltage((adc_channel_t)chan, &characteristics[attenuations[chan]], &mv);

  adc_result = mv * isr_float_t(1023) / isr_float_t(ADC_REFERENCE_VOLTAGE) / isr_float_t(1000);

```

bebu, 2022-06-30:
When you take a look at mks provided firmware they do this:
```
  // HAL_adc_result = mv * 1023.0 / 3300.0;
  HAL_adc_result = mv * 1023.0 / 2570.0;
```
I had similar problems with 2.5 so i set it to 2.57. Which is similar to what you have seen.

3DSmitty, 2022-06-30:
I ended up playing around with the value to get:

#define ADC_REFERENCE_VOLTAGE                  2.565

Setting it to 2.565 gives me +-1C degree accuracy.

thisiskeithb, 2022-06-30:
> Setting it to 2.565 gives me +-1C degree accuracy.

I've submitted a PR to update this: https://github.com/MarlinFirmware/Marlin/pull/24432