MarlinFirmware/Marlin #24145: [FR] Mechanical Gantry Calibration Current & Extra Height Adjustment in Marlin UI

sarvenn, 2022-05-07:
### Is your feature request related to a problem? Please describe.

While trying to find the best current and extra height I need to compile new fw which causes too much time loss. As far as I know there is not any GCODE command to send via a terminal to tune these parameters without the need of a new fw.

### Describe the feature you want

It would be nice to have a current and extra height (above z max pos) adjustment in the menu. 
Auto Z Align is the perfect place for it. It can be turned to a menu title and sub-menu could have execut auto align, current, extra height.

#define GANTRY_CALIBRATION_CURRENT          800 
#define GANTRY_CALIBRATION_EXTRA_HEIGHT       2

thisiskeithb, 2022-05-07:
There isn't a menu setup specifically for this, but you can do this with extra [`G34`](https://marlinfw.org/docs/gcode/G034-mgc.html) parameters: 

### Usage
`G34 [S<mA>] [Z<linear>]`

### Parameters
`[S<mA>]` Current value to use for the raise move. (Default: `GANTRY_CALIBRATION_CURRENT`)

`[Z<linear>]` Extra distance past Z_MAX_POS to move the Z axis. (Default: `GANTRY_CALIBRATION_EXTRA_HEIGHT`)

sarvenn, 2022-05-07:
> There isn't a menu setup specifically for this, but you can do this with extra [`G34`](https://marlinfw.org/docs/gcode/G034-mgc.html) parameters without needing to recompile:
> 
> ### Usage
> `G34 [S<mA>] [Z<linear>]`
> 
> ### Parameters
> `[S<mA>]` Current value to use for the raise move. (Default: `GANTRY_CALIBRATION_CURRENT`)
> 
> `[Z<linear>]` Extra distance past Z_MAX_POS to move the Z axis. (Default: `GANTRY_CALIBRATION_EXTRA_HEIGHT`)

My bad, I've missed this information: https://marlinfw.org/docs/gcode/G034-mgc.html
Thanks for the info @thisiskeithb .
Yeah, there is actually gcode commands to adjust these parameters but still that would be nice to have menu for it. It could be optional and could be enabled in configuration_adv.h such as //#define #define GANTRY_CALIBRATION_MENU

I hope someone would be interested to raise a PR for this.
I also plan to raise a ticket at BTT TFT repo to have these parameters adjustable in touch mode. I think it can be done without the need of any Marlin side update since the are gcode commands already, am I right?

thisiskeithb, 2022-05-07:
> I think it can be done without the need of any Marlin side update since the are gcode commands already, am I right?

Correct. Those TFTs are just computers that send/talk G-code.

sarvenn, 2022-05-14:
> `CUSTOM_MENU_MAIN` would also work for this:
> 
> https://github.com/MarlinFirmware/Marlin/blob/c1b53d63fc5a7fa5f17e5459e632cd8e7a0fd5ec/Marlin/Configuration_adv.h#L4002-L4018

Thanks for the tip @thisiskeithb.
Isn't it better to implement an Auto Z Align parameters sub-menu as a new feature which would be a nice improvement to Marlin? And I think it requires easy coding, just an HMI update, doesn't it?

thisiskeithb, 2022-05-14:
> And I think it requires easy coding

Easy is relative, but you can add `CUSTOM_MENU_MAIN` to your build without delay since it already exists within Marlin.

I’m just giving you an option if no one puts in a PR to fulfill this request since there’s [over 450 feature requests](https://github.com/MarlinFirmware/Marlin/issues?q=is%3Aopen+is%3Aissue+label%3A%22T%3A+Feature+Request%22) in the queue.