MarlinFirmware/Marlin #24149: [BUG] Inconsistent meaing if I-parameter for TMC gcode commands

rondlh, 2022-05-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I found some inconsistencies in the usage of the I-parameter for the TMC commands.

**M912** (clear TMC OTW) doesn't use the I-parameter at all (although the documentation says it does).

**M913** (Set TMC Hybrid Threshold Speed) and
**M914** (Set TMC stallguard sensitivity)
The meaning of the I-parameter is as follows:
No I-parameter or I0 means all drivers (X, X2)
I1 means driver 1 (X, Y, Z)
I2 means driver 2(X2, Y2, Z2)
The documentation is incorrect for both M913 and M914.

**M906** (set driver current) and
**M569** (Set spreadcycle or stealthchop)
The meaning of the I-parameter is as follows:
No I-parameter or I-1 means all drivers (X, X2)
I0 means driver 1 (X, Y, Z)
I1 means driver 2 (X, Y, Z)

### Bug Timeline

Don't know

### Expected behavior

Consistent use of I-parameter

### Actual behavior

Inconsistent use of I-parameter

### Steps to Reproduce

Check the source code...

### Version of Marlin Firmware

Marlin 2.0.9.3 latest bugfix (May 8th 2022)

### Printer model

Custom

### Electronics

Custom

hcet14, 2022-05-15:
Interesting issue. I'm planning to start with the TMCs soon. I wonder why nobody replied?

rondlh, 2022-05-15:
Good question, it seems not to be a hot topic. I'm working on implementing a TFT interface for Marlin, so a bit of consistency would be useful. 
From a previous discussion I understood that the M913/M914 convention is the correct one. (https://github.com/MarlinFirmware/Marlin/issues/24012)

Thanks for your support!

closed 2022-07-26 by github-actions[bot]