MarlinFirmware/Marlin #24150: ILI9488 SPI HOW TO IT WORK [FR] (feature summary)

minhngoc123, 2022-05-08:
### Is your feature request related to a problem? Please describe.

I USE BOARD BLACKSTM32F407VET6 CONNECTED TO LCD ILI9488 SPI BUT IT DOESN'T WORK
I TRY DISCONNECT the RESET pin and connect it to the 3.3V pin. SCREEN ON ONLY GRAY 1.
I TRY TRY #define TFT_ROTATION BUT EVERYTHING STILL DOESN'T WORK
IN FILE :
pins_BLACK_STM32F407VE.H I HAVE DEFINE LCD PAIN:
#define LCD_PINS_RS PE15
#define LCD_PINS_ENABLE PD8
#define LCD_PINS_D4 PE10
#define LCD_PINS_D5 PE12
#define LCD_PINS_D6 PD1
#define LCD_PINS_D7 PE8
#define BTN_ENC PD9
#define BTN_EN1 PD4
//#define BTN_EN2 PD13

#define DOGLCD_CS LCD_PINS_D5
#define DOGLCD_A0 LCD_PINS_D6

#define TFT_CS_PIN LCD_PINS_ENABLE
#define TFT_A0_PIN TFT_DC_PIN
#define TFT_SCK_PIN PB13 // SPI2_SCK
#define TFT_MISO_PIN PB14 // SPI2_MISO
#define TFT_MOSI_PIN PB15 // SPI2_MOSI
#define TFT_DC_PIN LCD_PINS_D7
#define TFT_RST_PIN LCD_PINS_RS

### Are you looking for hardware support?

BLACKSTM32F407VET6 VS ILI9488 SPI

### Describe the feature you want

i hope it work

### Additional context

[image]

ellensp, 2022-05-08:
You need to set up your screen, beyond just the pins..., I suspect you have TFT_GENERIC enabled,,,

See Marlin/src/inc/Conditionals_LCD.h
example for ILI9488

```
#elif ANY(MKS_ROBIN_TFT35, TFT_TRONXY_X5SA, ANYCUBIC_TFT35)                   // ILI9488
  #define TFT_DRIVER ILI9488
  #define TFT_DEFAULT_ORIENTATION (TFT_EXCHANGE_XY | TFT_INVERT_X | TFT_INVERT_Y)
  #define TFT_RES_480x320
  #define TFT_INTERFACE_FSMC

```

so you need to add a block something like this

```
#elif ENABLED(MY_CUSTOM_TFT)
  #define TFT_DRIVER ILI9488
  #define TFT_DEFAULT_ORIENTATION (TFT_EXCHANGE_XY | TFT_INVERT_X | TFT_INVERT_Y)
  #define TFT_RES_480x320
  #define TFT_INTERFACE_SPI

```
And then enable #define MY_CUSTOM_TFT instead of  TFT_GENERIC In your Configuration.h

minhngoc123, 2022-05-09:
> You need to set up your screen, beyond just the pins..., I suspect you have TFT_GENERIC enabled,,,
> 
> See Marlin/src/inc/Conditionals_LCD.h example for ILI9488
> 
> ```
> #elif ANY(MKS_ROBIN_TFT35, TFT_TRONXY_X5SA, ANYCUBIC_TFT35)                   // ILI9488
>   #define TFT_DRIVER ILI9488
>   #define TFT_DEFAULT_ORIENTATION (TFT_EXCHANGE_XY | TFT_INVERT_X | TFT_INVERT_Y)
>   #define TFT_RES_480x320
>   #define TFT_INTERFACE_FSMC
> ```
> 
> so you need to add a block something like this (with your correct resolution)
> 
> ```
> #elif ENABLED(MY_CUSTOM_TFT)
>   #define TFT_DRIVER ILI9488
>   #define TFT_DEFAULT_ORIENTATION (TFT_EXCHANGE_XY | TFT_INVERT_X | TFT_INVERT_Y)
>   #define TFT_RES_480x320
>   #define TFT_INTERFACE_SPI
> ```
> 
> And then enable #define MY_CUSTOM_TFT instead of TFT_GENERIC In your Configuration.h

i follow it but when buil it gives me an error that i have to choose 1 lcd screen while i already selected
[image]
[image]

ellensp, 2022-05-09:
You going to need to update the sanity check with your new display
Go to the line it fails on and add your new TFT to the list

jmz52, 2022-07-11:
ILI9488 and ILI9486 do not support 16-bit color when using SPI interface.
The only available options are 3-bit and 18-bit colors.
Marlin color UI was designed to work with 16-bit colors and STM32 hardware SPI interface with only works in 8-bit and 16-bit modes.

If you REALLY want to use ILI9488 with SPI interface you need transform each pixel's color from 16-bit format to 18-bit format before sending data to TFT and send _**3 full bytes**_ for each pixel. See page 120 of ILI9488 datasheet for exact bit order.
This will severely affect drawing performance and may affect overall firmware stability.