MarlinFirmware/Marlin #24152: Z_MIN_POS doesn't work in 2.0.9.3

mcer12, 2022-05-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

As the title says.. found out the hard way. To confirm this actually happend after homing, I auto-homed, then I set negative Z outside bed and again, it went ahead and moved to negative. This issue doesn't happen with X or Y. This is very bad because it actually can destroy the bed.

### Expected behavior

Assuming homing beforehand, with Z_MIN_POS set to 0, printer should prevent going below 0mm.

### Actual behavior

Z_MIN_POS is ignored

### Steps to Reproduce

1. auto home
2. set negative z
3. watch PEI sheet get destroyed

### Version of Marlin Firmware

2.0.9.3

### Printer model

Custom machine

### Electronics

SKR mini E3

### Bed Leveling

ABL Bilinear mesh

ellensp, 2022-05-09:
Please attach your configuration files.

mcer12, 2022-05-09:
Here it is:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8654936/Marlin.zip)

ellensp, 2022-05-09:
```
#define MIN_SOFTWARE_ENDSTOPS
#if ENABLED(MIN_SOFTWARE_ENDSTOPS)
  #define MIN_SOFTWARE_ENDSTOP_X
  #define MIN_SOFTWARE_ENDSTOP_Y
  //#define MIN_SOFTWARE_ENDSTOP_Z
```

You have it disabled

closed 2022-05-09 by ellensp