MarlinFirmware/Marlin #24153: [FR] (feature summary)

lanmgjy, 2022-05-10:
### Is your feature request related to a problem? Please describe.

We are trying to use stepping motor to drive lens instead of XY to try to replace galvanometer to create a project of laser sintering or laser resin curing, and request technical guidance or support from firmware. We know the conversion relationship between stepping angle and distance, but we are not able to modify the original driving conversion method

### Are you looking for hardware support?

No need at present. We have built relevant hardware platforms

### Describe the feature you want

//提供机器高度，绝对坐标，电机一圈的脉冲数，返回一个距离对应振镜角度的脉冲数
float sls(float Hgh,float length,float pull)
{
float angle;
angle =(pull/365)*tan(Hgh/length) ;
return(angle);
}

This is a dynamic formula we expected

### Additional context

I hope to provide some help in Marlin kinematics modification

thinkyhead, 2022-08-20:
It's true that kinematics in Marlin is a bit scattered throughout the code base and there is no one "kinematics" class where it all comes together. We had a submission once for Galvanometer control but I never got around to integrating that feature, probably because it was too difficult for me at that time. Maybe we can take another stab.

Generally, Marlin's motion control assumes stepper motors with STEP / ENA / DIR signals, so to do motion with another type of device can require adjustments. However, if your lens angling is being done with steppers then you just have to follow the model of Delta or SCARA kinematics, in which lines are divided up into very small segments to minimize error in non-linear kinematics.

To get more help with this feature you should join our Discord server and there you should be able to find a collaborator or two who can help you to put this together.

thinkyhead, 2022-08-20:
It's true that kinematics in Marlin is a bit scattered throughout the code base and there is no one "kinematics" class where it all comes together. We had a submission once for Galvanometer control but I never got around to integrating that feature, probably because it was too difficult for me at that time. Maybe we can take another stab.

Generally, Marlin's motion control assumes stepper motors with STEP / ENA / DIR signals, so to do motion with another type of device can require adjustments. However, if your lens angling is being done with steppers then you just have to follow the model of Delta or SCARA kinematics, in which lines are divided up into very small segments to minimize error in non-linear kinematics.

To get more help with this feature you should join our Discord server and there you should be able to find a collaborator or two who can help you to put this together.

- Marlin Discord server join link: https://discord.gg/n5NJ59y