MarlinFirmware/Marlin #24154: Ender 3 V2 Jyers UI: The "Use probe" option when manual bed leveling is not working as should.

TravieRulezd00d, 2022-05-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When trying to use the "Use Probe" option, the probe will only probe the right 2 sides during any stage of the operation. It also doesn't read out any measurements that you should adjust to, it always says 0.

### Bug Timeline

I first tried JyersUI in December, and it happened then aswell

### Expected behavior

To probe all 4 corners at the beginning of the operation, then allow me to probe every corner to see measurements for leveling.

### Actual behavior

It'll only probe both right corners. Adjustment value always reads 0 on every corner, even the ones that successfully probe.

### Steps to Reproduce

Just attempt to use "Use Probe" when using Manual Bed Leveling

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Ender 3 V2

### Electronics

Stock electronics

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

tititopher68-dev, 2022-05-10:
@TravieRulezd00d 
Absolutely not a code issue ! You have an hardaware issue.
look at : https://e1.pcloud.link/publink/show?code=XZlrUzZIjqdIXXN63X7MuvH5wLDVHLMolj7

tititopher68-dev2, 2022-05-10:
@TravieRulezd00d 
I don't know why..But you can try to compile the firmware using the configuration files present on Jyers' GitHub, (Configuration.h / Configuration_adv.h) or at least comparing them to yours.

thisiskeithb, 2022-05-11:
We can’t support forks here. 

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

TravieRulezd00d, 2022-05-11:
@thisiskeithb 
This was indeed the latest bugfix branch. I first started experiencing this issue back in December.

tititopher68-dev2, 2022-05-11:
> @thisiskeithb This was indeed the latest bugfix branch. I first started experiencing this issue back in December.

This is what I understood, but as soon as one evokes something other than their GitHub, Marlin said no.  While perhaps the outcome would be due to their configuration files that are not always functional and suitable for UI.  As the Ender3v2, there are 4 different UI...

TravieRulezd00d, 2022-05-21:
@tititopher68-dev2
So I took your advice and tried compiling with the Jyers configuration files, but to no avail unfortunately. It actually makes it even less far into the process, it doesn't try to probe at all now. Idk but I must be doing something wrong when compiling.

ellensp, 2022-07-07:
JyersUI has been removed from Marlin.  See https://github.com/MarlinFirmware/Marlin/pull/24459 for details.

closed 2022-07-07 by ellensp