MarlinFirmware/Marlin #24155: [FR] SPI_ENDSTOPS for Multi-Stepper / Multi-Endstop

rondlh, 2022-05-10:
### Is your feature request related to a problem? Please describe.

I have a printer with 2 Y-steppers driven by 2 TMC5160 drivers, homing works fine using the diag pins, but homing completely fails when using SPI_ENDSTOPS. (SPI communication is established and reported OK).
I have the same issue for and IDEX printer, homing the X carriages completely fails when enabling SPI_ENDSTOPS.

I had a quick look at the source code, which makes me think this is not a bug, but the capability is just not implemented. I gave it a try myself, but failed.
By using SPI_ENDSTOPS I can free up some hardware endstops connectors, which will allow me to implement new features, like a nozzle probe, filament runout detection and LED support.

### Are you looking for hardware support?

Nope

### Describe the feature you want

SPI_ENDSTOPS support for Multi-Stepper / Multi-Endstop systems, specifically:
Dual Y-driver, dual Y-endstop
IDEX X-carriage.

### Additional context

Freeing up the endstops allows me to implement new/more features.

rondlh, 2022-11-08:
Not sure if is a good solution, but it works for me.
In motion.cpp expand the #if Y_SPI_SENSORLESS like shown:
```
    #if Y_SPI_SENSORLESS
      if (tmc_spi_homing.y && (stepperY.test_stall_status()
        #if ANY(CORE_IS_XY, MARKFORGED_XY, MARKFORGED_YX) && X_SPI_SENSORLESS
          || stepperX.test_stall_status()
        #elif CORE_IS_YZ && Z_SPI_SENSORLESS
          || stepperZ.test_stall_status()
        #endif
      )) {
        SBI(live_state, Y_ENDSTOP);
		hit = true;
      }  
      #ifdef Y_DUAL_ENDSTOPS
	  if (tmc_spi_homing.y && (stepperY2.test_stall_status() // Added section for dual endstop Y
        #if ANY(CORE_IS_XY, MARKFORGED_XY, MARKFORGED_YX) && X_SPI_SENSORLESS
          || stepperX.test_stall_status()
        #elif CORE_IS_YZ && Z_SPI_SENSORLESS
          || stepperZ.test_stall_status()
        #endif
      )) {
        SBI(live_state, Y_ENDSTOP + 2);
		hit = true;
      } 
      #endif	  
    #endif
```

A few lines below add this in void Endstops::clear_endstop_state()
```
	#ifdef Y_DUAL_ENDSTOPS
      TERN_(Y_SPI_SENSORLESS, CBI(live_state, Y_ENDSTOP + 2)); // Added to clear Y2 endstop status
    #endif
```

This does the trick for me. It's not nice to use Y_ENDSTOP + 2, but Y2_ENDSTOP doesn't work.

thisiskeithb, 2023-02-12:
Added in https://github.com/MarlinFirmware/Marlin/pull/25371

closed 2023-02-12 by thisiskeithb