MarlinFirmware/Marlin #24156: [BUG] Arduino DUE + RAMPS_FD_V2 keeps restarting

scott0122, 2022-05-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Arduino DUE keeps restarting no matter what RAMPS_FD_V2 is installed or not.

Only 2 options are changed in default configuration.h

#define MOTHERBOARD BOARD_RAMPS_FD_V2
#define BAUDRATE 115200

### Version of Marlin Firmware

Latest bugfix-2.0.x

ej0rge, 2022-05-10:
I seem to be experiencing the same issue with my RADDS.

Bob-the-Kuhn, 2022-05-12:
I'm also getting a periodic reset.  Pronterface reports it as a WDT timeout.

I disabled the WDT and re-downloaded.  Not it just burps out the powerup messages and just sits there.

---

I had forgotten how much "fun" downloading a DUE can be.  

---

I remember trying to find a V2 a couple of years ago.  I ended up butchering a V1 to get the V2 functionality and to get TMC SPI daisy chain drivers working.  Sure glad the STM32 boards are now relatively affordable.

ej0rge, 2022-05-12:
In my case it is a reboot loop on boot since updating past 2.0.9.3. When i get a round tuit i will put some time and effort into figuring out which update to that tree broke it. 

Dunno what you mean about DUE fun. octoprint plugs into the native port, vscode plugs into the programming port by a long cable from a desktop machine here. Click build click upload. On a previous DUE board i removed the FETs from both usb ports so that the board can't be powered via USB, but i let the magic smoke out of that one.

Bob-the-Kuhn, 2022-05-13:
Windows and the DUE download can be a pain on my PC.  Sometimes it's nice, sometimes it's not.

I was able to attach my J-Link to a DUE and get it into the failing mode (with WDT disabled).  It's been a couple of years since I looked at this but as best I can tell it's running the main command loop.  It hits the breakpoints at the idle() and queue.advance() multiple times.

thisiskeithb, 2022-05-13:
> When i get a round tuit i will put some time and effort into figuring out which update to that tree broke it.

This would be a good candidate: 
- https://github.com/MarlinFirmware/Marlin/pull/23357

ellensp, 2022-05-13:
Confirmed it was https://github.com/MarlinFirmware/Marlin/pull/23357
If you git checkout fee85b31  (previous commit to 23357) it works as expected
If you git checkout 44eff9a2 (the 23357 commit) It reboots with Watchdog Reset

ellensp, 2022-05-13:
also DUE_debug build environment is listed in pins.h all over, but does not exist....

from older versions

```
[env:DUE_debug]
# Used when WATCHDOG_RESET_MANUAL is enabled
platform    = atmelsam
extends     = env:DUE
build_flags = ${common.build_flags}
  -funwind-tables
  -mpoke-function-name

```

marciot, 2022-05-16:
Please check whether PR #24182 addresses this issue.

Bob-the-Kuhn, 2022-05-16:
Didn't fix it.

I grabbed the latest bugfix, set the motherboard, compiled & downloaded, verified that the WDT problem exists, changed the one file, compiled & downloaded and still had the same problem.

Just for grins & giggles I tried the other two combinations and still had the same result - spits out about 1K characters and then stops until the WDT kicks in. 

Whatever is killing it is doing so before all the power up characters are sent out. Sends most but not all.

Bob-the-Kuhn, 2022-05-20:
Looks like the problem is fixed.  Just downloaded bugfix, set the motherboard, compiled & downloaded.  Communicates just fine with no WDT issues.

Don't understand why my attempt to reproduce the PR #24182 changes didn't succeed 4 days ago.  Just glad that all is well now.

thisiskeithb, 2022-05-20:
Closing since https://github.com/MarlinFirmware/Marlin/pull/24182 has been merged & problem is resolved.

closed 2022-05-20 by thisiskeithb