MarlinFirmware/Marlin #24159: [BUG] Cannot enable PREHEAT_BEFORE_LEVELING with PROBE_MANUALLY

jonasmalacofilho, 2022-05-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to enable `PREHEAT_BEFORE_LEVELING` with `AUTO_BED_LEVELING_BILINEAR` and `PROBE_MANUALLY`.

However, this fails to compile:

```
Marlin/src/gcode/bedlevel/abl/G29.cpp: In static member function 'static void GcodeSuite::G29()':
Marlin/src/gcode/bedlevel/abl/G29.cpp:440:32: error: 'class Probe' has no member named 'preheat_for_probing'
         if (!abl.dryrun) probe.preheat_for_probing(LEVELING_NOZZLE_TEMP,
                                ^~~~~~~~~~~~~~~~~~~
*** [.pio/build/mega2560/src/src/gcode/bedlevel/abl/G29.cpp.o] Error 1
```

I initially found this issue while trying to compile Marlin 2.0.9.3 for a Creality Ender-5 Pro, but I can also reproduce it with `bugfix-2.0.x` @ 4a17589496d1 and minimal changes from the default (RAMPS_14_EFB) configuration: simultaneously enabling `AUTO_BED_LEVELING_BILINEAR`, `PROBE_MANUALLY`, and `PREHEAT_BEFORE_LEVELING`.

The immediate cause is that `preheat_for_probing` is only defined if `HAS_BED_PROBE`, and `PROBE_MANUALLY` doesn't count.  But I think that it should (in this specific instance).

### Bug Timeline

Unsure

### Expected behavior

Compilation to succeed, as preheating the nozzle and bed seems sensible even when manually probing.

Alternatively, an informative error message, in the style of the sanity checks, explaining why this is not a good idea.

### Actual behavior

Compilation failed.

### Steps to Reproduce

1. Simultaneously enable `AUTO_BED_LEVELING_BILINEAR`, `PROBE_MANUALLY`, and `PREHEAT_BEFORE_LEVELING`
2. Build

### Version of Marlin Firmware

`bugfix-2.0.x` @ 4a17589496d1 (alternatively: 2.0.9.3 release)

### Printer model

Default configuration (alternatively: Creality Ender-5 Pro with V4.2.2 board)

### Electronics

Everything stock

### Add-ons

Everything stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Additional information & file uploads

This issue can be reproduced on `bugfix-2.0.x` @ 4a17589496d1 with the default (RAMPS_14_EFB) configuration and minimal changes:

```patch
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 875adc21a481..686c8339d2a4 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -1150,7 +1150,7 @@
  * Use G29 repeatedly, adjusting the Z height at each point with movement commands
  * or (with LCD_BED_LEVELING) the LCD controller.
  */
-//#define PROBE_MANUALLY
+#define PROBE_MANUALLY
 
 /**
  * A Fix-Mounted Probe either doesn't deploy or needs manual deployment.
@@ -1666,7 +1666,7 @@
  */
 //#define AUTO_BED_LEVELING_3POINT
 //#define AUTO_BED_LEVELING_LINEAR
-//#define AUTO_BED_LEVELING_BILINEAR
+#define AUTO_BED_LEVELING_BILINEAR
 //#define AUTO_BED_LEVELING_UBL
 //#define MESH_BED_LEVELING
 
@@ -1681,7 +1681,7 @@
 /**
  * Auto-leveling needs preheating
  */
-//#define PREHEAT_BEFORE_LEVELING
+#define PREHEAT_BEFORE_LEVELING
 #if ENABLED(PREHEAT_BEFORE_LEVELING)
   #define LEVELING_NOZZLE_TEMP 120   // (°C) Only applies to E0 at this time
   #define LEVELING_BED_TEMP     50
```

---

Alternatively/in reality I was trying to compile Marlin 2.0.9.3 with [this configuration (zip)](https://github.com/MarlinFirmware/Marlin/files/8672983/matilda-config-e344c1ca3579.zip).

ellensp, 2022-05-11:
PROBE_MANUALLY is not considered to be a probe, so probe object and its functions are not enabled.

a4jp-com, 2025-04-14:
I want this feature as when we manually check the bed height we should be allowed to do it with the bed and nozzle heat set to the correct temperature. Even if we are only using the nozzle to check the correct height.

closed 2025-04-16 by thinkyhead