MarlinFirmware/Marlin #24161: [BUG] (bug summary)

microsoul, 2022-05-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

TRIGORILLA 14  i3 mega anycube.
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:880:0,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:35,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins_postprocess.h:791:8: error: #error "No E stepper plug left for Y2!"
       #error "No E stepper plug left for Y2!"
        ^~~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins_postprocess.h:877:8: error: #error "No E stepper plug left for Z2!"
       #error "No E stepper plug left for Z2!"

### Bug Timeline

new

### Version of Marlin Firmware

bugfix-2.0.x STRING_DISTRIBUTION_DATE "2022-05-12"

### Printer model

Anycube i3 mega

### Additional information & file uploads

error in compilation

thisiskeithb, 2022-05-12:
The Trigorilla 14 board only has five driver slots, so you can't enable options for non-existent driver slots. You can do Y2 or Z2, but not both.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

microsoul, 2022-05-12:
in Configuration.h in config/examples\AnyCubic\i3Mega
define X2_DRIVER_TYPE ALL_DRIVERS_TYPE
define Y2_DRIVER_TYPE ALL_DRIVERS_TYPE
define Z2_DRIVER_TYPE ALL_DRIVERS_TYPE
is not decomented for default

thisiskeithb, 2022-05-12:
This issue should be transferred to the Configurations repo, but it looks like those driver types were enabled by mistake since `X_DUAL_STEPPER_DRIVERS` and `Y_DUAL_STEPPER_DRIVERS` were disabled by default, but recent changes to Marlin now uses enabled `*_DRIVER_TYPE`s to set those features.

Comment out/disable `X2_DRIVER_TYPE` and `Y2_DRIVER_TYPE`.

I've put in https://github.com/MarlinFirmware/Configurations/pull/734 to fix the config.

microsoul, 2022-05-14:
i have this:
#define Z2_DRIVER_TYPE ALL_DRIVERS_TYPE

but motor z2 does not move

i dont find setting:

#define NUM_Z_STEPPER_DRIVERS

microsoul, 2022-05-14:
I did other tests:
with the command:
G0 Z22
both motors move perfectly

with command:
G28 Z
it moves only on a motor

microsoul, 2022-05-15:
Solved
Configurations/config/examples/AnyCubic/i3 Mega/Configuration.h

Line 974
#define Z_MIN_ENDSTOP_INVERTING true  // Set to true to invert the logic of the endstop.

thisiskeithb, 2022-05-16:
@microsoul: Are you running a completely stock AnyCubic i3 Mega?

microsoul, 2022-05-16:
Yes

thisiskeithb, 2022-05-16:
> Yes

I’ve included this fix in https://github.com/MarlinFirmware/Configurations/pull/737